prusa3d/PrusaSlicer #5144: Seam editing/painting on multi-part models

tg73, 2020-11-12:
### Version

2.3.0-alpha3+win64

### Operating system type + version
Win10 x64

### Behavior
* _Describe the problem_
I can't select one part of a multi-part model to paint seams. The seam painting tool is only active when a root part is selected. I need to paint seams on a face of part of a multi-part model that is contained within another part.

For example, I want to have the seams on the red part where I've outlined in blue:

[image]

_Is this a new feature request?_
I don't know

#### Project File (.3MF) where problem occurs
Sorry, I can't share this part. It's a common scenario in any multi-part model. I can mock something up if required.

bubnikv, 2020-12-07:
IMHO we need a different strategy for hiding seams on multi-material objects. We need to hide the seams automatically at the inner walls.

bubnikv, 2021-09-22:
https://github.com/prusa3d/PrusaSlicer/pull/6751

bubnikv, 2022-06-29:
Implemented in PrusaSlicer 2.5.0-alpha2. Closing.