prusa3d/PrusaSlicer #5145: [Feature Request] Hole Horizontal Compensation

konehead, 2020-11-12:
New feature request:

I'm a convert from Cura to PrusaSlicer and I really love it.
One of the only things I really miss is the Cura feature called "Hole Horizontal Expansion". 
I was very excited when this was recently added to Cura and it seems to work quite well.

Basically it's similar to the PrusaSlicer feature called "XY Size Compensation", but it only affects internal polygons (therefore not affecting external dimensions of the model).

This would give us another tool to help correct this very common issue with FDM printing (internal dimensions being too small)

See also: Issue #515 

Thanks,
Jeff

lonelymyp, 2020-11-15:
The first requests to implement such a function appeared 5-7 years ago.
The author of the slic3r then said that these are not problems of the slicer, but of the material and mechanics, it does not need to be solved by software.

I would very much like this problem to be finally solved.

Markus-1984, 2020-12-28:
I would like it, too

readconsult, 2020-12-30:
Add my vote.

elascent, 2021-02-10:
Yep, the problem is very common. +1

spiky2021, 2021-02-13:
Some of our engineers ask me to find a solution for this problem, as well.
They simply don't wont to store several designs for the same product in their database.
The risk of mistaking is too high.  
In general a solution already exists for a while in the SuperSlicer branch of PS and Cura does 
it similar. 
In the old discussion about the sense of a slicer based workarounds I completly agree with the 
argument of the PS responsibles, that it is a material problem. But the same discussion concluded 
as well, that it is difficultly solved by a direct material parameter correlation. Even if it is found, its 
use will still suffer from missing information provided by material manufactures.
A similar situation exists in the injection molder business.
Therefore I enter into the above request to implement a similar solution like the one of SS to PS,
but as separate material profile parameters.

rtyr, 2021-02-25:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4561. Closing.

closed 2021-02-25 by rtyr

NVNDO, 2022-01-29:
I just swapped to the newest  PS version and as this feature is missing I'm back to Cura. What do I need a Slicer for if it doesn't allow me to print an object in accurate dimensions? ...and I was seriously considering to buy a Prusa-machine in the future - seems like that´s not going to happen. 

PD: As nobody from Prusa even replies to this, its clear to see  that they work mainly in their own interest and not ours. 
Conclusion: PrusaSlicer: A nice and awesome piece of Software, but sadly USELESS, unless you just want to print toys. 

A real shame...

rajhlinux, 2023-08-03:
> The first requests to implement such a function appeared 5-7 years ago. The author of the slic3r then said that these are not problems of the slicer, but of the material and mechanics, it does not need to be solved by software.
> 
> I would very much like this problem to be finally solved.

Can you explain more on this?