prusa3d/PrusaSlicer #5148: Won't upload .gcode files to Octoprint if there is a custom processing script

dsiguero, 2020-11-13:
### Version
2.2.0+

### Operating system type + version
macOS 10.13.6 (17G65)

### 3D printer brand / version + firmware version (if known)
Voron 2.4 running Klipper

### Behavior
I have recently started using a Python post-processing script to be able to remove the default nozzle heat gcode (I'm handling that with a Klipper macro, on a per-filament basis). The script just removes certain lines at the beginning of the file, as can be seen below. I basically adapted the example [here](http://projects.ttlexceeded.com/3dprinting_prusaslicer_post-processing.html).

To my surprise, I wasn't able to upload to Octoprint (v1.4.2) directly from PrusaSlicer anymore, getting the following error:

```
Error uploading to print host:
HTTP 411: <html><title>411: Length Required</title><body>411: Length Required</body></html>
```

If I disable the post-processing script (emptying the `Post-processing scripts` text-area in `Print Settings > Output options`), the Octoprint upload works successfully.

#### Post-processing script

```python
#!/usr/bin/python
import sys
import re
import os

sourceFile=sys.argv[1]

# Read the ENTIRE g-code file into memory
with open(sourceFile, "r") as f:
    lines = f.readlines()

destFile = re.sub('\.gcode$', '', sourceFile)
os.rename(sourceFile, destFile + ".no_init_voron.bak")
destFile = re.sub('\.gcode$', '', sourceFile)
destFile = destFile + '.gcode'

search_string = 'filament-specific start g-code'
flag_found = False

with open(destFile, "w") as of:
    for lIndex in xrange(len(lines)):
        oline = lines[lIndex]
        
        if not flag_found:
            if (oline.startswith(';') or len(oline.strip()) == 0) and not search_string in oline:
                of.write(oline)
            else:
                flag_found = search_string in oline
        else:
            of.write(oline)
        
of.close()
f.close()
```

Kind regards

bubnikv, 2020-12-08:
Aren't you creating an empty file with your script?
https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/411

dsiguero, 2020-12-08:
> Aren't you creating an empty file with your script?
> https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/411

Nope, I am currently uploading the result gcode file to Octoprint. And they print fine.

FidelCapo, 2020-12-10:
I tested the upload with the script and the upload was ok.

Script:
`#!/bin/bash
echo "Press any key to continue"
while [ true ] ; do
read -t 3 -n 1
if [ $? = 0 ] ; then
exit ;
else
echo "waiting for the keypress"
fi
done`

dsiguero, 2020-12-10:
> I tested the upload with the script and the upload was ok.
> 
> Script:
> `#!/bin/bash echo "Press any key to continue" while [ true ] ; do read -t 3 -n 1 if [ $? = 0 ] ; then exit ; else echo "waiting for the keypress" fi done`

Not sure whether this is relevant considering it doesn't change the original gcode file. It doesn't prove the upload still works after modifying the gcode.

FidelCapo, 2020-12-11:
With this script is upload also ok.
`#!/bin/bash
echo "Press any key to continue"
while [ true ] ; do
read -t 3 -n 1
if [ $? = 0 ] ; then
exit ;
else
echo "waiting for the keypress"
fi
echo "this gcode was changed" > $1
done`

dsiguero, 2020-12-14:
> With this script is upload also ok.
> `#!/bin/bash echo "Press any key to continue" while [ true ] ; do read -t 3 -n 1 if [ $? = 0 ] ; then exit ; else echo "waiting for the keypress" fi echo "this gcode was changed" > $1 done`

You're right, there was a problem when creating the output file. It now works using the first argument as the destination file.

closed 2020-12-14 by dsiguero