prusa3d/PrusaSlicer #5150: [2.3.0-alpha3] Seam painting/editing is not respected by internal perimeters

tg73, 2020-11-13:
### Version
2.3.0-alpha3

### Operating system type + version
Win10 x64

### Behavior

Painting seam enforcers only affects external perimeters. Internal perimeters do not respect seam painting, even when using sphere. Under certain circumstances, the seam on internal perimeters shows through to the outer perimeter. Ideally, all perimeters should respect seam positions. I'm aware of the "External perimeters first" setting, but this has other side effects, so is not an ideal workaround.

In this photo, the enforced seam is outlined blue, the seams from internal perimeters show through where outlined green:
[image]

You can tell from looking at retractions and unretractions in the sliced model where this problem will occur. I can't share the actual model from the photo, attached is a simple project showing the problem.

Note: reproducing the printed result shown above will depend upon filament selection, temperature, etc. There may be good reasons for selecting materials and other configuration that tends to produce blobby seams, and just to be clear, that's not the issue I'm concerned about. I'm saying this up front to avoid suggestions about how to avoid blobby seams.

From the attached test project, enforced seam is outlined blue, internal perimeters not respecting seam position is outlined green:

[image]

#### Project File (.3MF) where problem occurs

[Seam painting is not respected by internal perimeters.zip](https://github.com/prusa3d/PrusaSlicer/files/5536548/Seam.painting.is.not.respected.by.internal.perimeters.zip)

fe60, 2020-11-13:
@tg73 Do I miss something or is this issue a copy of #5149?

tg73, 2020-11-13:
GitHub was being weird. I got a 404 the first time I clicked submit. I didn't see it had made a dupe. I'll close this one!

closed 2020-11-13 by tg73