prusa3d/PrusaSlicer #5151: [FR] Label objects default on

paukstelis, 2020-11-13:
I'm putting the suggestion out there to have the Label Objects tick box be turned on by default in all standard profiles. It does not add many additional lines to gcode. It provides nice markers for those that have to look at gcode manually and would also save a lot of grief when it comes to Octoprint-CancelObjects plugin when people don't realize (or forget) to turn this on when using standard profiles.

fe60, 2020-11-16:
Found this in the Prusa Forum:
>`gcode_label_objects = 1`
>In the PrusaResearch.ini file, there are sections that inherit other sections, so I just added the line under
>`[print:*MK3*]`
>and it's now in the default for every layer and speed option I have. 

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/how-to-change-default-profiles-in-prusaslicer/#post-267253

paukstelis, 2020-12-08:
> Found this in the Prusa Forum:
>
>
> https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/how-to-change-default-profiles-in-prusaslicer/#post-267253

Those also no longer seems to work in 2.3.0. :(

combs, 2021-01-01:
This should absolutely be the default. I'd be really surprised if enough people had the MMU2 to warrant having this option off by default.

rtyr, 2021-01-05:
It will be enabled in the next configuration update (most likely this week). Closing.

closed 2021-01-05 by rtyr