prusa3d/PrusaSlicer #5153: Adding a Repetier Server Physical Printer with 2 printers crashes Prussa Slicer

gjpc, 2020-11-13:
### Version
2.3.0-alpha3+win64

### Operating system type + version
Win 64 Version 10.0.19041 Build 19041

### 3D printer brand / version + firmware version (if known)
Repetier server with a Prusa MK3S and Custom Zonestar with BTT SKR 1.4 Marlin

### Behavior
Prusa Slicer crashes to desktop when the the refresh printers button is pressed in the Physical Printer dialog

* _Steps needed to reproduce the problem_
1. install repetier server add two printers
2. run Prussa Slicer V2.3.0-alpha3-win64
3. add Prussa MK3S printer
4. add Custom Printer Zonestar BTT SKR 1.4
5. Select Pinter settings tab
6. Select Edit Physical printer icon
7. set name to Rep
8. Select Repetier in Host Type Drop down
9. type 127.0.0.1:3344 in Hostname field
10. Fill in the API Key field
11. Select Test button and confirm Connection
12 Pressing the Refresh Printers button causes Prussa Slicer to crash to desktop

* _Expected Results_
See list of printers???

* _Actual Results_
crash to desktop

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
NA

mloidl, 2020-11-14:
I can confirm this behavior.
I've tracked down the crash to 
```
    Choice *choice = dynamic_cast<Choice*>(rs);
    choice->set_values(printers); // here it crashes
    printers.empty() ? rs->disable() : rs->enable();
```
in PhysicalPrinterDialog.cpp

bubnikv, 2020-11-21:
fixed by @ mloidl

closed 2020-11-21 by bubnikv

bubnikv, 2020-11-30:
Merged now after all.

gjpc, 2020-11-30:
This exact same sequence crashes Beta 4. Please re-open.

lukasmatena, 2020-11-30:
@gjpc 
There is no need to reopen, the fix was merged after beta1.

gjpc, 2020-11-30:
Oh Sorry, I'll re-test beta 2 when it is released.  Thanks lukasmatena

gjpc, 2020-12-10:
Everything works very smoothly on beta 2, I was even able to print to the server directly from the Slicer. The job does not auto-start but I think in the long run that is a plus. Kudos @bubnikv  & @lukasmatena

PR #5159 (2020-11-30): Possible segmentation fault in physical printer dialog when Host is set to "Repetier"
When pressing "Refresh Printers" in Physical Printer Dialog slicer crashed because of a wrong cast

this will fix #5153