prusa3d/PrusaSlicer #5154: [2.3.0-alpha3+win64] - hangs while preparing infill

Trellian, 2020-11-14:
### Version
2.3.0-alpha3+win64

### Operating system type + version
Windows 10 Home 64 bit, Acer Aspire
### 3D printer brand / version + firmware version (if known)
Stock Geeetech A10M

### Behavior
Hangs while preparing infill, regardless of whether auto-slice is on or not

* _Expected Results_
No Hanging
Hanging, eventual high CPU usage

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Pro

Can not upload the .3MF file directly, as requested above... so I zipped it 
[Fain Rain Cover - V2 - part 1 of 2.zip](https://github.com/prusa3d/PrusaSlicer/files/5540329/Fain.Rain.Cover.-.V2.-.part.1.of.2.zip)

FidelCapo, 2020-11-25:
I cannot reproduce this issue. Can you please try slice this 3mf with 2.3.0-alpha4+win64 version? Thank you.

Trellian, 2020-11-25:
Thanks. The issue seems to have disappeared in 2.3.0-alpha4-win64

closed 2020-11-25 by Trellian