prusa3d/PrusaSlicer #5160: Preview window 'view' setting changes on iterative slicing

NielschB, 2020-11-14:
### Version
PrusaSlicer 2.3.0-alpha3 (Also observed in 2.2.0)

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S + MMU2S

### Behavior
When slicing an object for the first time after prusaslicer startup, the preview is shown with its view selection set to 'Feature type'. When slicing the model again, the view in the preview is set to 'Tool'.
I would expect the view setting to stay at the last selected value, or at least at the default value (Feature type) for iterative slices.

mkoistinen, 2020-11-22:
Can confirm this on MacOS v11 (Big Sur) too. It constantly changes itself to "Tool", regardless of where I left it before the slice operation. This did not happen in previous releases.

FidelCapo, 2020-12-08:
This issue should be fixed in 2.3.0-beta2
Closing.

closed 2020-12-08 by FidelCapo