prusa3d/PrusaSlicer #5162: Feature Requst: Object tree is too small for MMU/object manipulation

Area5142, 2020-11-15:
### Version
PrusaSlicer-2.3.0-alpha3 and previous alphas

### Operating system type + version
I use Linux Mint 19.3 x64 but all platforms is affected.

### 3D printer brand / version + firmware version (if known)
All

### Behavior
When assigning colors to MMU print or manipulation modifiers I use the _object tree list_ in the side panel. The working area of this list has decreased in size for every release of PrusaSlicer as the sidebar functionality has grown:

[image]

Working with this small list is very annoying as it changes size whenever there is a change (background processing on) and the element in focus often disappears outside the visible window.

I find no alternative to object extruder assignment besides this list.
When the model has many parts or the build plate has many parts it is easy to loose the overview and spend a lot of time trying to find the correct object to change.
Using long names forces the horizontal scroll bar to be used as the sidebar has a max width, further complicating navigation.

### Suggested solutions
Best solution would be to make the sidebar tabbed with the _object-tree_ in one of the tabs.

A smaller change to the current sidebar could be optimizing it, giving more space for the individual elements:
1. Make section: _Print Settings_, _Filament_, _Printer+Support+Infill_, _Group Manipulation_, _Object-tree_ and _Sliced Info_ into collapsible sections. This would make space for all the non-collapsed elements. Make the setting persistent between sessions.
2. Make Object-tree user sizable, but without auto sizing as today. This would make the list easier to work with, not jumping around in size.
3. Move _Inches_ checkbox to its rightful place in settings making more space. Normally a user works in either Imperial or Metric system.

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
No project/gcode needed

bubnikv, 2020-12-12:
The object list has been extended in PrusaSlicer 2.3.0-beta3.

lukasmatena, 2020-12-18:
After complains that the object list is now too large (#5462), we reduced (8849aa77dc5fe4eacaafd29c274dcbdacc638aef) the height again so the minimum is now larger than originally but not as high as after the first change. We will likely fix this eventually by changing the clamping limits dynamically based on the number of items, but not for 2.3.0 release.

Area5142, 2020-12-19:
When no objects is selected, the _tree view_ has a usable size, but when an object is selected the _Instance manipulation_ panel pops up and and "steals" the space.
Unfortunately it is not possible to change an object without selecting it, like when changing extruder/color.

If somehow the _Instance manipulation_ panel could collapse or hide it would provide the necessary space for a bigger tree view. It could be as simple as clicking the heading label _Instance manipulation_ and toggle it on/off. It already has the code to hide when no object is selected.

closed 2026-09-29 by prusaslicer-bot[bot]