prusa3d/PrusaSlicer #5163: [2.3.0-alpha3] Text in drop-down fields on Linux is misaligned

timur-tabi, 2020-11-15:
### Version
[2.3.0-alpha3]

### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
MK3s

### Behavior
Most of the drop-down fields have misaligned text.

[image]

bubnikv, 2020-12-07:
@yusanka did quite some adjustments of the Linux UI layouts. Would you please test with PrusaSlicer 2.3.0-beta2? Thanks.

timur-tabi, 2020-12-07:
It's fixed in PrusaSlicer-2.3.0-beta2+linux-x64-202012051122.AppImage.  It's still broken in PrusaSlicer-2.3.0-beta2+linux-x64-GTK3-202012051553.AppImage however.  I'm assuming the latter is the "GTK" version of PrusaSlicer.

bubnikv, 2021-01-06:
We have received a hint on wx-users list to try to set the following attribute at the GTK combo box widget. @tamasmeszaros will try that.

```
g_object_set( G_OBJECT( cell ), "ellipsize", PANGO_ELLIPSIZE_END, NULL ); 
```

timur-tabi, 2021-04-20:
Just FYI, this problem still exists in 2.3.1.  As before, it's only with the GTK version.

bubnikv, 2021-07-30:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/a2b337ee4543f7fbbde07ca5b929baf9da410dbf
The fix will be released with PrusaSlicer 2.4.0-alpha1.

closed 2021-07-30 by bubnikv