prusa3d/PrusaSlicer #5165: Feature request: Support for .pwmx format

nospoon, 2020-11-15:
Could you please add support for the Anycubic Photon Mono X file format .pwmx? The slicer Anycubic provide is unusable, but it is the only one that can generate these files. All the Photon Mono X users desperately need an alternative DLP slicer.

ravibowman, 2020-12-24:
I would also love to see this happen. I love PrusaSlicer for FDM Printing and now own an AnyCubic Mono X. Would love to be able to use it for both printers :).

abdirk, 2021-02-12:
I'd like to join the request. Please add support for pwmx. The slicer from Anycubic is buggy and there are no alternatives.

ICH88, 2021-08-17:
Yes please!

bubnikv, 2021-09-22:
We don't have the resources to support Anycubic.

progman32, 2021-12-21:
Just as an FYI for folks wanting to use PrusaSlicer for their Photon. I have a mostly-ok workflow where I export the STL with supports from PrusaSlicer (I made a custom profile so the build plate is sized correctly), then import the generated STL into Photon's slicer (in linux under wine).

closed 2022-02-14 by lukasmatena

progman32, 2022-02-14:
This is the best Valentine's Day gift. Thank you, @ole00, and 3 cheers for open workflows.