prusa3d/PrusaSlicer #5166: Top layer missing

gde-2, 2020-11-15:
### Version
_Version of PrusaSlicer used goes here_
Used 2.2.0+ linux (docker keyglitch/prusaslicer:2.2.0 and 2.3.0-rc1 keyglitch/prusaslicer)
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Docker version

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Creality - Ender 3

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
My design is rather simple: a coin with a flat button and on the top I have a border and a relief inside the coin
The slicer does a great job but once it gets to the top layer it completly misses the intended design

_Is this a new feature request?_
NO
#### Project File (.3MF) where problem occurs
_U
[Design.zip](https://github.com/prusa3d/PrusaSlicer/files/5542773/Design.zip)
pload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-11-15:
Those missing parts are too thin to be printed with default extrusion width. Either reduce the extrusion width in advanced settings (0.40 seems ok) or enable the "Detect thin walls" option. Or modify your design to be printable with your default settings.

gde-2, 2020-11-15:
Okay thank you. Never thought about this.

closed 2020-11-15 by gde-2