prusa3d/PrusaSlicer #5167: Utilisation de PRUSASCLICER V2.3.0

JCD95-lab, 2020-11-15:
### Version
_Version of PrusaSlicer used goes here_2.3.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ windows 7
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ Artillery Genius

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
Bonjour, je suis un novice sur prusaslicer et la version 2.3.0 m'apporte globalement une réponse à mes attentes pour les supports.
la vidéo de démonstration est très bien faire et est très utile.
J'ai deux petites remarques :
_ lorsque l'on récupère un fichier STL réalisé en pouce, il s'en rend compte mais ne fait pas ce qu'il faut ou à l'envers, ce qui fait que l'on obtient un objet 25.4 fois plus petit (il faut répondre NON, les unités affichées sont alors en inc et la tailles est correcte). En fait il n'y a pas de conversion....
_ l'utilisation de la peinture fonction très bien; sauf si dans l'objet il y a des rectangles très fin, là la seule solution est de prendre un cercle très grand (pas top car beaucoup de sélections non désirées), il faudrait pouvoir utiliser la fenêtre générée par le déplacement de la souris pour sélectionner.

Une question : est-il prévu des supports en arbres ?

Si joint le projet sur lequel j'ai travaillé (PEGASE récupéré sur Thingivers)
[PEGASUS3.zip](https://github.com/prusa3d/PrusaSlicer/files/5542775/PEGASUS3.zip)

Cordailement JC. DERBAY

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-11-22:
Could you please describe the issue in English?

bubnikv, 2020-11-24:
Google translate

Hello, I am a novice on prusaslicer and version 2.3.0 brings me overall an answer to my expectations for supports.
the demo video is doing very well and is very helpful.
I have two small remarks:
_ when we recover an STL file made in inches, it realizes it but does not do the right thing or backwards, which means that we obtain an object 25.4 times smaller (it must answer NO, the displayed units are then in inc and the size is correct). In fact there is no conversion ....
_ the use of paint works very well; unless in the object there are very thin rectangles, there the only solution is to take a very large circle (not great because many unwanted selections), it would be necessary to be able to use the window generated by the movement of the mouse to to select.

One question: are tree supports planned?

If attached the project I worked on (PEGASE retrieved from Thingivers)
PEGASUS3.zip

Cordailement JC. DERBAY

bubnikv, 2020-11-24:
I cannot make sense of the automatic translation. No response, closing.

closed 2020-11-24 by bubnikv