prusa3d/PrusaSlicer #5168: Crash with printer profile containing a "print_host" setting

smurfix, 2020-11-15:
### Version

master cd2b502

### Operating system type + version

Linux Debian Testing

### 3D printer brand / version + firmware version (if known)

Prusa, doesn't matter

### Behavior

* add non-empty `print_host = XX` setting to your default printer's config file
* start slicer
* crash

```
Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x00007fff00000001 in ?? ()
(gdb) whe
#0  0x00007fff00000001 in  ()
#1  0x0000555557bde060 in  ()
#2  0x0000555557bde2c0 in  ()
#3  0x0000555557bde060 in  ()
#4  0x00007ffff4c35b38 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#5  0x00007ffff45c1429 in g_object_run_dispose () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#6  0x00007ffff698a288 in wxWindow::~wxWindow() ()
    at /usr/lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#7  0x00005555556b5bdd in wxMessageDialog::~wxMessageDialog()
    (this=0x7fffffff91b0, __in_chrg=<optimized out>) at /usr/include/wx-3.0/wx/gtk/msgdlg.h:14
#8  0x00005555556b5bdd in Slic3r::GUI::GUI_App::check_printer_presets() (this=0x555556648000)
    at ./src/slic3r/GUI/GUI_App.cpp:1155
#9  0x0000555555bd5860 in Slic3r::GUI::GUI_App::load_current_presets() (this=this@entry=0x555556648000)
    at ./src/slic3r/GUI/GUI_App.cpp:1764
#10 0x0000555555bda9a8 in Slic3r::GUI::GUI_App::on_init_inner() (this=0x555556648000)
    at ./src/slic3r/GUI/GUI_App.cpp:970
#11 0x0000555555bdbda6 in Slic3r::GUI::GUI_App::OnInit() (this=<optimized out>)
    at ./src/slic3r/GUI/GUI_App.cpp:747
#12 0x00007ffff6e6fa72 in wxEntry(int&, wchar_t**) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#13 0x0000555555bcf702 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...)
    at ./src/slic3r/GUI/GUI_Init.cpp:81
#14 0x000055555572ccac in Slic3r::CLI::run(int, char**)
    (this=<optimized out>, argc=<optimized out>, argv=<optimized out>) at ./src/PrusaSlicer.cpp:567
#15 0x000055555570c82d in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /usr/include/c++/8/ext/new_allocator.h:79
```

bubnikv, 2020-11-24:
Would you please provide your data to test with? Thanks.

bubnikv, 2020-12-07:
No response, no data, cannot reproduce.
Closing.

bubnikv, 2020-12-07:
well, I see a crash in splash screen after an exception is thrown, catched, an error message is displayed and closed.

closed 2026-09-29 by prusaslicer-bot[bot]