prusa3d/PrusaSlicer #5171: Start G-code

Dane809, 2020-11-16:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-11-16:
Please add some description.

Dane809, 2020-11-16:
Hey there,

here are the Start-Gcode whicht throwing the error.

`G28 ; home all axes
M140 S[first_layer_bed_temperature] ;set bed temperature
M104 T[current_extruder] S[first_layer_temperature]
M109 T[current_extruder] S[first_layer_temperature]
M190 S[first_layer_bed_temperature] ;wait for bed temperature
{if T[current_extruder]==T[0]} G92 E0; Extruder 0 setzen
{if T[current_extruder]==T[0]} G1 E10 F300 ; Extruder 10mm foerdern
{if T[current_extruder]==T[0]} G1 Z5 F800
{if T[current_extruder]==T[0]} G1 X70 Y60 F1500;
{if T[current_extruder]==T[0]} G92 E0
{if T[current_extruder]==T[0]} G1 Z0.2
{if T[current_extruder]==T[0]} G1 X150 E8 F800; Initial Line
{if T[current_extruder]==T[0]} G92 E0
{endif}
{if T[current_extruder]==T[1]}  G92 E0; Extruder 0 setzen
{if T[current_extruder]==T[1]}  G1 E10 F300 ; Extruder 10mm foerdern
{if T[current_extruder]==T[1]}  G1 Z5 F800
{if T[current_extruder]==T[1]}  G1 X280 Y60 F1500;
{if T[current_extruder]==T[1]}  G92 E0
{if T[current_extruder]==T[1]}  G1 Z0.2
{if T[current_extruder]==T[1]}  G1 X200 E8 F800; Initial Line
{if T[current_extruder]==T[1]}  G92 E0
{endif}`

Dane809, 2020-11-16:
which description do you mean?

neophyl, 2020-11-16:
Please correct me if I’m wrong but it was my understanding that current extruder can’t be used in start gcode as at that point it doesn’t know what the current extruder is.

Dane809, 2020-11-16:
M104 T[current_extruder] works perfectly. 
I think the current_extruder can´t be use at an if statement.

Dane809, 2020-11-18:
Are there any news?

foreachthing, 2020-11-18:
Maybe, just maybe try it this way:
```
{if current_extruder==0}
; code for Extruder 0
G92 E0; Extruder 0 setzen; # COMMENT: this line doesn't make sense to me. 
G1 E10 F300 ; Extruder 10mm foerdern
; and so on...
{elsif current_extruder==1}
; code for Extruder 1
{else}; not Extruder 0 or 1{endif}
```

I did not test it with multiple extruders.

neophyl, 2020-11-18:
I just tried it with 2 extruders and yep that seems to work nicely.

Dane809, 2020-11-18:
Okay thank you so much guys 😉

closed 2020-11-18 by Dane809