prusa3d/PrusaSlicer #5174: Add print sheet selection or type placeholder in Prusa Slicer

GWdd, 2020-11-16:
### Version
2.3.0.a.2

### Operating system type + version
Win10x64

### 3D printer brand / version + firmware version (if known)
Prusa Mk3/s 3.9.1+

### Behavior
When writing gcode - as far as I can discover - there is no way to modify a print to accommodate printing characteristics of different print sheets.

Yes, a new profile can be made for each sheet type, but that seems a little bit onerous. A simple tick mark in the printer profile would suffice. 

In Printer Settings, General, under bed shape add BED TYPE [textured/smooth] to add layer one changes like higher layer one extrusion rate to better fill sheet texture and make layer 1 perform better.

M221 S{if sheet_type==textured && layer_num==0}120{else}95{endif}

This would be an expert level change, as it can also cause problems.  

Or, this would be even better if the control were in the printer: when a textured sheet is selected a layer one compensation number can be entered: Sheet:[Textur1] Layer0Comp:1.20

But since firmware space is limited... at slice is next best.

_Is this a new feature request?_
YES

GWdd, 2020-11-16:
Must be an easier way to add the label ENHANCEMENT

bubnikv, 2020-11-24:
Please check the content of the Printer settings notes. This is the section copied from the Mini printer:

Don't remove the following keywords! These keywords are used in the "compatible printer" condition of the print and filament profiles to link the particular print and filament profiles to this printer profile.
PRINTER_VENDOR_PRUSA3D
PRINTER_MODEL_MK3

This is the "Compatible printers condition" for the 0.2mm QUALITY Print settings:

printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK3.*/ and nozzle_diameter[0]==0.4

You see there is a check for a keyword PRINTER_MODEL_MK3 in the printer_notes field.
You may add your keywords for the print bed type there and use the regular expression test into your custom G-codes.

GWdd, 2020-11-24:
No clue what I was thinking ... but it looks like a separate Print profile per sheet type is what is needed.

closed 2020-11-24 by GWdd