prusa3d/PrusaSlicer #5175: Add instances and objects print order please

ZhenyaKa, 2020-11-17:
Please add order of instances or objects like them orderd in list. Now the printed almost randomly.

For example I need print 16 cylinders and need mode "Complete individual object"

My head not round and I need print order from left-near coner to lefr far coner and next line right one step
^
| 4  7 ...
| 3  8 ...
| 2  6 ...
| 1  5 ...
--------------->

After pressing Arrange button I receive printing 4 objects in center and other around theese 4pcs.

Please arrange instances at plate like them ordered in objects list and enable possiblity to sort them in that list. Now I cant move instances in that list.

Objects print order also havn't any logics and not linked with list position

foreachthing, 2020-11-17:
My print head with gantry is not round either. I miss this setting, like cura has, where one can set that up correctly.
I don't like to add an R 62 mm, when my print head in Y only measures 30 mm (in total!!) => lots of wasted space.

But, why does the order matter? You still need the same clearance? Am I missing something?

If you can't move your parts around, try to disable "Auto-center parts) in preferences [CTRL]+[P].

ZhenyaKa, 2020-11-17:
My head not round too and will be fine to ajust distance from nozzle center in all four direction.
I can move parts, but printing order very weird.

This order I need for printing small nylon parts.

bubnikv, 2020-11-21:
> Please arrange instances at plate like them ordered in objects list and enable possiblity to sort them in that list. Now I cant move instances in that list.

you can drag and drop the objects in the object list to reorder them. Objects are printed in that order.
You will also see the order by pressing the "e" hotkey in the platter.

ZhenyaKa, 2020-11-22:
I can't reoder instances. Only objects I can.

bubnikv, 2020-11-23:
> I can't reoder instances. Only objects I can.

That is true. For now you would have to drag & drop the instance outside of
the object at the object list to make it an independent object.

quetzalcoatl, 2021-02-03:
@bubnikv On PrusaSlicer 2.3.0 I just found out that reordering objects on the right-side panel doesn't cause the slicing algorithm to obey the new order. It still does what it wants.

I'm testing quality of my manual-bed-levelling setup. For that reason, I'm printing a series of very flat squares. My I suspect some leaking/stringing problem, so I want to change the direction relatively often, so that squares are not printed row-by-row or column-by-column. Using the list on the right-side panel, I ordered them this way:

[image]

However, after slicing, when I inspected the print order of the first layer, here's what I got:

[image]

Here's the whole printing project in 3MF file, it should contain all my current settings. I suppose you should be able to simply open it, press slice and see the difference. (if anyone finds it helpful for calibration/testing, feel free to use it on MIT license).

[test_squares_x25_270x270.printing.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5917859/test_squares_x25_270x270.printing.3mf.zip)

keeena, 2021-03-16:
I'd also like to have the ability to order instances. Currently (v2.3.0) you can order objects (different objects, including copies of an object), but you cannot order instances of an object.

The workaround of using copies instead of instances isn't useful if/when you have to modify the item. Copies requires you to make the change to all, but using instances will change all of them at once (which makes instances much more useful).

ochm, 2021-11-05:
In current 2.4.0-beta1 the print in order of list of object does not work. I have named object and sorted them from top (01) to down (10) and placed them to be printed counter clockwise, start from left down, end left up.
But the real print (if i use horizontal slider to animate layer print progress) seems to be done from top right, then down right, etc. So IMHO the sort order or naming does not make sense. I'm not printing in sequential mode, just want to optimize path of printing, but it is not possible now. All can be checked in attached project.
[cedulka2-color.zip](https://github.com/prusa3d/PrusaSlicer/files/7489163/cedulka2-color.zip)
.

af12689, 2021-11-20:
I tested the object ordering in 2.4.0-beta1 and found that the object order as defined by the object list is only honored if "Complete individual objects" (CIO) is enabled.

The attached screenshot shows the objects "0".."9" quasi randomly positioned on the build plate. With CIO enabled the object "0" is printed first, then "1", then "2", and so on, just like it is defined in the object list.
[image]

If CIO is disabled, then (per layer) the print order is a zig-zagging line starting at the top left ("1", "0", "4", "7", "9", "2", "3", "5", "8", "6"). The object order in the object list is ignored.
[image]

In non-CIO mode the object order should also be honored.

ochm, 2021-11-20:
> If CIO is disabled, then (per layer) the print order is a zig-zagging line starting at the top left ("1", "0", "4", "7", "9", "2", "3", "5", "8", "6"). The object order in the object list is ignored. 

Yes, and there is long travel from object 6 to object 1 in next layer. Manually sorted objects can be in order where the first and the last object are near to each other, then no long travel is needed.

bubnikv, 2021-12-02:
@quetzalcoatl 

Your project has sequential print order disabled. If I enable it, I get an error report that the objects are too close, which is newly visualized with PrusaSlicer 2.4.0-beta2.

[image]

If I decrease extruder clearance, then the project slices with sequential print order enabled and the print order correctly matches the order in the object list.

quetzalcoatl, 2021-12-07:
@bubnikv Hmm.. indeed I didn't pay attention to clearance, since the squares are very flat and there's no risk of any collisions. I also did not expect it to have any impact on ordering, but the sequential-print, yes, that's more likely it. That's odd though, I'm pretty sure I had it checked. I'll see at the original files that went to printer once I get there and if they're still there. Thank you very much!

MrDixioner, 2022-02-03:
> However the automatic order is quite optimal.
It's not optimal. For example, 9 pieces of objects are printed. It starts at the top right and ends at the bottom left. And then makes a long way back to the first object. This causes incomplete printing, especially on flowable plastics such as PETG.
We need to make sure that the first and last objects are printed as close to each other as possible, and not at a distance from the earth to the sun.
[image]

amishne, 2023-09-29:
Upvoting: Being able to fine-tune the order of objects probably has many other benefits beyond optimizing travel time, in particular for multicolor/multimaterial material printing like the Prusa XL. Two I can immediately think of:

1. You can ensure that the nozzle doesn't pass over other objects, which otherwise needlessly introduces risk of oozing into them (especially bad if they have different colors) or colliding into them.
2. You can manually create prime/purge towers, which can be useful in some scenarios - for example if you are printing with PETG and PLA and you want a separate prime tower for each.

jonatasfreitasv, 2024-01-04:
Any update about this feature?

sslupsky, 2024-10-03:
I would like to see ability to override the auto order and manually choose the order of instances.

MrDixioner, 2025-05-21:
When will it be possible to manually select the print order? This topic is still open. Is it really that hard to do or do the developers not want it?
I have 6 parts. I have arranged them in order, but the printing is done in a completely different way.
It is necessary to make it possible to select parts in the 3D window and assign them a print order.
[image]

sslupsky, 2026-09-02:
Still need the ability to set the order in which the instances are printed.