prusa3d/PrusaSlicer #5177: Enabling Avoid Crossing Perimeters Takes Unreasonably Long To Slice

mmcglumphy, 2020-11-17:
### Version
2.3.0-alpha3

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3s MMU2s

### Behavior
If I enable Avoid Crossing Perimeters in Print Settings, slicing takes a very long time. 

I routinely print 24h+ prints and find this feature to result in excellent quality to the final print. The Slic3r feature is either buggy or needs optimization. Cura can Slice with this enabled and generally take minutes or less. With Slic3r, I can leave it for over an hour before it finishes and many times it hangs and I cannot cancel and must end task.

bubnikv, 2020-11-17:
We are revamping this feature for PrusaSlicer 2.3.0

bubnikv, 2020-11-17:
See also 
https://github.com/prusa3d/PrusaSlicer/labels/path%20planning

hejllukas, 2020-11-17:
The avoid crossing perimeters after the revamp should be faster than the old one in many cases. Could you please provide 3MF or STL file, which takes very long to slice with the avoid crossing perimeters enabled? We can look at it and test it with the revamped version.

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

closed 2020-12-11 by bubnikv

mmcglumphy, 2020-12-11:
> "Avoid crossing perimeters" has been reworked with [5f456eb](https://github.com/prusa3d/PrusaSlicer/commit/5f456ebbe78f3b994ea7ecc7de23c119e4705b61) (see the branch lh_avoid_crossing_perimeters).
> The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

Awesome! Will try it in the next beta and report any issues.