prusa3d/PrusaSlicer #5181: Feature request: ABL grid per print

theflu, 2020-11-17:
Not sure if this is possible with gcode but I think it would be nice if the slicer could generate ABL probing points based off the object that is being printed. That way you could way there could be a more accurate mesh of the area that is being printed on.

bubnikv, 2020-11-24:
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha1

The G-code placeholder parser supports the following new keys: "print_bed_min", "print_bed_max", "print_bed_size", which are calculated as a bounding box of an existing "bed_shape" vector of points.

The G-code placeholder parser also newly supports the following new keys: "first_layer_print_convex_hull", "first_layer_print_min", "first_layer_print_max", "first_layer_print_size", defining convex hull of the first layer. Please note that the extrusions generated by the custom G-code are not accounted for.

You should be able to script whatever you need with these new placeholders.

miniskipper, 2020-11-28:
G28 ;Home
G29  L{first_layer_print_min[0]}  R{first_layer_print_max[0]}  F{first_layer_print_min[1]}  B{first_layer_print_max[1]}   ;ABL within print dimensions

Bilinear Auto Bed Level  on Marlin 2.x

:)

bubnikv, 2020-11-28:
So you could close this issue?

closed 2020-11-28 by theflu