prusa3d/PrusaSlicer #5182: OSX Green Text Is Unreadable

alsmithson, 2020-11-18:
### Version
2.30 Alpha 2+ and Alpha 3+

### Operating system type + version
OSX 10.14.6 Mohave

### 3D printer brand / version + firmware version (if known)
Not relevant

### Behavior
Open the slicer and click the "Printer Settings" tab. The green text on the printer settings screen is lighter than it was in PrusaSlicer 2.2.0+ and it is placed on a lighter background than 2.2.0+. This new color scheme is impossible to read on the light gray background, and very hard to read on the white background of the sidebar (see screenshots). Please return the green text to its previous color.

[image]
[image]

neophyl, 2020-11-18:
Is this related to https://github.com/prusa3d/PrusaSlicer/issues/4679 ?

lukasmatena, 2020-11-20:
@neophyl It almost certainly is. I would actually consider it to be a duplicate. Something must have been broken in between 2.2.0 and now. Although it does not appear everytime.

@FidelCapo Are you able to reproduce it somehow on your system?

alsmithson, 2020-11-20:
It is certainly related. But it isn't just background color, because the white in the sidebar was always white. It is the green that has changed. My search did not bring up #4679 before posting this, or I would have added my comment there, my apologies.

bubnikv, 2020-12-07:
I don't remember the details, but I believe that the dark mode support in OSX Mohave is half baked and it is really best to be disabled. I am not sure whether it is worth investing our time into it, as the OSX Mohave is quite old.

I would recommend to disable dark mode in PrusaSlicer as described here:
https://github.com/prusa3d/PrusaSlicer/issues/4679#issuecomment-731285989

closed 2026-09-29 by prusaslicer-bot[bot]