prusa3d/PrusaSlicer #5183: [2.3.0-alpha3] - Extrusion width not as per input parameters

jeblaze, 2020-11-18:
### Version
2.3.0-alpha3

### Operating system type + version
Windows 10 Pro 20H2 x64

### 3D printer brand / version + firmware version
Creality CR-10S Pro V2 with Mosquito Hot-End - Using Marlin 2.6.1 with DGUS reloaded touchscreen firmware

### Behavior
* With identical slicing parameters between PS 2.2.0 and 2.3.0-alpha3, PS 2.3.0-alpha3 seems to generate way thinner lines than input parameters, prints are also as per slicing result.
* See attached config.ini generated from PS 2.3.0-alpha3 in the .ZIP file
* Should be the same as PS 2.2.0
* Actual Results (for those who wonder, it's a drop-in Mosquito adapter for the CR-10S Pro that I designed) 
  * PS 2.3.0-alpha3, extrusion width 0.4mm
[image]
  * PS 2.2.0, extrusion width 0.4mm
[image]

#### Project File (.3MF) where problem occurs
See .ZIP file.

#### Attached files
[PS_2.3.0-alpha3.zip](https://github.com/prusa3d/PrusaSlicer/files/5557744/PS_2.3.0-alpha3.zip)

rtyr, 2020-11-18:
Your extrusion width is set to 0.12mm, not 0.4mm. I don't see any issue here.

[image]

rtyr, 2020-11-20:
No response. Closing.

closed 2020-11-20 by rtyr