prusa3d/PrusaSlicer #5188: Send gcode to octorpint triggers firewall [lulu]

zuhra031, 2020-11-18:
### Version
PrusaSlicer-2.2.0+-202003211132

### Operating system type + version
Mac OS 10.14.6

### 3D printer brand / version + firmware version (if known)
DIY - marlin 2.0.7.2

### Behavior
I have slicer connected with octoprint (octopi). Sometimes when I hit "send to printer" LuLu (firewall) ask for permission to allow PrusaSlicer process to access network (every time I hit allow [permanent] in LuLu)

Its not a bug but it knows to be annoying - sometimes I don't see LuLu window and send to printer option throws error "timeout".

[image]

bubnikv, 2020-12-03:
What is LuLu?

What do you advise?

zuhra031, 2020-12-03:
@bubnikv 
LuLu - firewall app for MacOS. https://objective-see.com/products/lulu.html
Been using beta version last few days. Every thing is ok

bubnikv, 2020-12-03:
Isn't this a LuLu issue then?

zuhra031, 2020-12-03:
@bubnikv 
I believe its not LuLu. LuLu work as it should - it asks for every process on system is it ok to access the network (lan/wan).

I'm thinking its problem with slicer/part of slicer that is responsible for gcode deployment on octoprint. It makes (on random as far as I noticed) temp processes on system which it uses to deploy file.

As mentioned in earlier post, I'm using v2.3.0-beta1 and it works fine, haven't found a bug so far :).

bubnikv, 2020-12-04:
> It makes (on random as far as I noticed) temp processes on system which
it uses to deploy file.

It launches threads, not processes to transfer a file over the network.

I have no idea of how to convince LuLu to ask just once. Would you please
check with your LuLu provider?

closed 2026-09-29 by prusaslicer-bot[bot]