prusa3d/PrusaSlicer #5189: [Bug] Weird Times-Roman font due to bad API call

daleckystepan, 2020-11-18:
### Version
2.3.0 commit 72f9cc07e80b80b131378148a195f7cef004a6d6

### Operating system type + version
MacOS Big Sur

### 3D printer brand / version + firmware version (if known)
Not relevant

### Behavior
Times-Roman font due to bad API call. You can see it for example on the Slice button.

#### Project File (.3MF) where problem occurs
Not relevant

➜  build git:(master) src/prusa-slicer
2020-11-18 15:23:58.104 prusa-slicer[59179:3260750] CoreText note: Client requested name ".SFNS-Regular", it will get Times-Roman rather than the intended font. All system UI font access should be through proper APIs such as CTFontCreateUIFontForLanguage() or +[NSFont systemFontOfSize:].
2020-11-18 15:23:58.104 prusa-slicer[59179:3260750] CoreText note: Set a breakpoint on CTFontLogSystemFontNameRequest to debug.

[image]

FidelCapo, 2020-11-19:
You are right. We know about this issue. It is new with Big Sur OS. Thanks for report and hint.

mkoistinen, 2020-11-22:
Kudos to the @daleckystepan for finding a possible culprit, but this issue existed even before https://github.com/prusa3d/PrusaSlicer/commit/72f9cc07e80b80b131378148a195f7cef004a6d6, please peep this screen shot under Big Sur:
[image]

Notice the Times New Roman in the "Slice now" button already.

daleckystepan, 2020-11-23:
Yes, this issues probably exists even before mentioned commit. I just noticed it and report it so I picked current commit I was using.

bubnikv, 2020-11-28:
Fixed by patching our wxWidgets.
https://github.com/prusa3d/wxWidgets/commit/5ceb8d834b2b80d54fea132bde7ae388ba61cea2
https://github.com/prusa3d/wxWidgets/commit/cd9f1404f674720192274ddb7ed743e34aeffa0d

It will be available as part of the PrusaSlicer 2.3.0-beta1 release.

closed 2020-11-28 by bubnikv