prusa3d/PrusaSlicer #5190: [2.3.0-alpha3] Arrow Up/Down doesn't work in Preview, while U/D still do

fe60, 2020-11-18:
### Version
PrusaSlicer-2.3.0-alpha3+win64-202011041552

### Operating system type + version
Windows 10

### Behavior
* Import an arbitrary object, slice it and open the layers preview
* Click somewhere on the preview (just to ensure it has the focus)
* press `Arrow Down`
_Expected Result_
The number of shown layers is decreased (as in PrusaSlicer 2.2)
_Actual Result_
Nothing happens

You can still use `U` or `L` to change the number of layers shown, but `Arrow Up/Down` only works, when the focus is on the layer slider, but it doesn't while the focus is on the preview.

_Is this a new feature request?_
No

fe60, 2020-12-05:
Arrow keys are functional again in 2.3.0-beta2 (incl. Arrow Left/Right).
U/D became W/S.

closed 2020-12-05 by fe60

foreachthing, 2020-12-05:
And A/D is for tool path (or gcode line) navigation 👍

bubnikv, 2020-12-06:
> And A/D is for tool path (or gcode line) navigation

Or left / right arrow if you are inside the 3D scene.
and with Ctrl or Shift key it jumps with larger steps.

We will make our G-code viewer the best G-code viewer in the world :-)