prusa3d/PrusaSlicer #5191: Crash (segmentation fault) when deleting all copies of a model

xhalo32, 2020-11-18:
### Version

`PrusaSlicer-2.3.0-alpha2+linux-x64-202010241644.AppImage`

### Operating system type + version

Arch Linux `5.9.6-arch1-1`

### 3D printer brand / version + firmware version (if known)

Not relevant

### Behavior

When adding a model (for example, right click -> box) and instances of it, slicing them and deleting them from the Plater causes PrusaSlicer to crash. It is very unpredictable but adding more instances usually makes it more likely to crash.

Steps to reproduce:
1. Move `.config/PrusaSlicer-alpha` to another location
1. Start PrusaSlicer and cancel the wizard (resetting the slicer every time is not necessary, the steps after this can be repeated for the crash)
1. Right click in the plater view and add a box shape
1. Add 1 instance with the + button (expert mode is not required even though the toolbar doesn't show the 'Add instance' tool)
1. Select all with Ctrl-A and press Delete to remove them
1. Press Ctrl-Z
1. Press Delete again (every instance should be selected)
1. An error dialog pops up saying Last instance of an object cannot be deleted

The crash can also occur without ever undoing with Ctrl-Z (after step 5), but I only managed to get that to happen with actual settings, but in that case there is no error dialog.

If I save the project file after step 6, the program doesn't crash, but the error dialog still appears.

Console logs from PrusaSlicer:
```
...
(prusa-slicer:911856): GLib-GObject-WARNING **: 17:53:06.181: invalid (NULL) pointer instance

(prusa-slicer:911856): GLib-GObject-CRITICAL **: 17:53:06.181: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:911856): GLib-GObject-WARNING **: 17:53:06.181: invalid (NULL) pointer instance

(prusa-slicer:911856): GLib-GObject-CRITICAL **: 17:53:06.181: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
[1]    911856 segmentation fault (core dumped)  ./PrusaSlicer-2.3.0-alpha3+linux-x64-202011041552.AppImage
```
> The console is actually filled with hundreds of those warnings

This is what the error dialog says:
[image]

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
I could not create a project file which would cause the issue on the first delete. Pressing delete in the imported project file (exported after step 6) does not cause a crash, but if I undo and delete again, the same crash happens.

> Background information: The original situation where the bug happened: I added 5 copies of a small object and then deleted them after they were sliced (background processing was enabled). If you want the original files and configuration I can provide them.

lukasmatena, 2020-11-20:
Thanks. There is indeed a problem of some sort. I have a build with ASAN and that crashes just after Ctrl+A and Delete. No undo-redo is needed. This is what it says:
<details>
<summary>ASAN report</summary>
<code>
    ================================================================
    
    ==6116==ERROR: AddressSanitizer: heap-use-after-free on address 0x6160003705e8 at pc 0x000000a8afd4 bp 0x7ffe4be20060 sp 0x7ffe4be20050
    READ of size 8 at 0x6160003705e8 thread T0 (slic3r_main) 

    #0 0xa8afd3 in Slic3r::ModelObject::get_object_stl_stats() const (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0xa8afd3)
    #1 0xa8c275 in Slic3r::ModelObject::get_mesh_errors_count(int) const (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0xa8c275)
    #2 0x253c075 in Slic3r::GUI::ObjectList::get_mesh_errors_list(int, int) const (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x253c075)
    #3 0x254156b in Slic3r::GUI::ObjectList::part_selection_changed() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x254156b)
    #4 0x255589c in Slic3r::GUI::ObjectList::selection_changed() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x255589c)
    #5 0x2571e04 in wxEventFunctorFunctor<wxEventTypeTag<wxDataViewEvent>, Slic3r::GUI::ObjectList::ObjectList(wxWindow*)::{lambda(wxDataViewEvent&)#1}>::operator()(wxEvtHandler*, wxEvent&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2571e04)
    #6 0x4aba2fa in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:671
    #7 0x4b5a4d9 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1415
    #8 0x4b5b1a6 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1887
    #9 0x4b5a921 in wxEvtHandler::TryHereOnly(wxEvent&) /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1608
    #10 0x4b5bd44 in wxEvtHandler::TryBeforeAndHere(wxEvent&) /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/event.h:3912
    #11 0x4b5a770 in wxEvtHandler::ProcessEventLocally(wxEvent&) /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1545
    #12 0x4b5a707 in wxEvtHandler::ProcessEvent(wxEvent&) /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1518
    #13 0x4b5aa1c in wxEvtHandler::SafelyProcessEvent(wxEvent&) /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1636
    #14 0x495f0ff in wxWindowBase::HandleWindowEvent(wxEvent&) const /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/wincmn.cpp:1555
    #15 0x4a68de9 in wxdataview_selection_changed_callback /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/dataview.cpp:4503
    #16 0x7fd25127b1d3  (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x101d3)
    #17 0x7fd2512959a5 in g_signal_emit_valist (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x2a9a5)
    #18 0x7fd251296554 in g_signal_emit_by_name (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x2b554)
    #19 0x7fd24fe7a366  (/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0+0x226366)
    #20 0x7fd25127afa4 in g_closure_invoke (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0xffa4)
    #21 0x7fd25128cfc0  (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x21fc0)
    #22 0x7fd251295d5b in g_signal_emit_valist (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x2ad5b)
    #23 0x7fd25129608e in g_signal_emit (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x2b08e)
    #24 0x4a608bc in wxGtkDataViewModelNotifier::ItemDeleted(wxDataViewItem const&, wxDataViewItem const&) /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/dataview.cpp:1834
    #25 0x49a4aae in wxDataViewModel::ItemDeleted(wxDataViewItem const&, wxDataViewItem const&) /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/datavcmn.cpp:170
    #26 0x2a56cb4 in Slic3r::GUI::ObjectDataViewModel::Delete(wxDataViewItem const&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2a56cb4)
    #27 0x2a54f39 in Slic3r::GUI::ObjectDataViewModel::Delete(wxDataViewItem const&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2a54f39)
    #28 0x2a55a59 in Slic3r::GUI::ObjectDataViewModel::Delete(wxDataViewItem const&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2a55a59)
    #29 0x2545554 in Slic3r::GUI::ObjectList::delete_object_from_list(unsigned long) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2545554)
    #30 0x2545bc6 in Slic3r::GUI::ObjectList::delete_from_model_and_list(Slic3r::GUI::ItemType, int, int) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2545bc6)
    #31 0x2ed3da8 in Slic3r::GUI::Selection::erase() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2ed3da8)
    #32 0x23b8564 in Slic3r::GUI::Plater::remove_selected() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x23b8564)
[... cut ...]
    #15 0x491c92f in wxMenuBase::SendEvent(int, int) /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/menucmn.cpp:643
    #16 0x4a3f377 in menuitem_activate /home/lukas/Slic3r/Slic3r/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/menu.cpp:541
    #17 0x7fd25127afa4 in g_closure_invoke (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0xffa4)

    SUMMARY: AddressSanitizer: heap-use-after-free (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0xa8afd3) in Slic3r::ModelObject::get_object_stl_stats() const
</code>
</details>

YuSanka, 2020-12-04:
Thanks for your report. This bug is fixed and will be a part of upcoming release

closed 2020-12-04 by YuSanka