prusa3d/PrusaSlicer #5192: [2.3.0-alpha3] Notification "Exporting finished" does not fade out.

fe60, 2020-11-18:
### Version
PrusaSlicer-2.3.0-alpha3+win64-202011041552

### Operating system type + version
Windows 10

### Behavior
* _Steps needed to reproduce the problem_
* Slice model
* Export g-code
* _Expected Results_
Notification about export appears and fades out after a while.
* _Actual Results_
Notification about export appears and stays forever, when not closed by the user.

Please consider making the export notification fading out. As it offers buttons/links for further action, the time until it fades out could be longer compared to "normal" notifications.

_Is this a new feature request?_
No.

The announcement about the new _Notification center_ in the release notes of [2.3.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha1) mentions notifications about warnings and errors will stay as long as they are valid.
The validity of the link offered in the standard notification and the eject button shown in the notification after an export to a removable media depends on the way a user works. I use the short cut `Ctrl + T` to eject the media. When doing an export to a local folder, I usually switch to a file explorer to copy the newly generated file to another location. In both workflows the notification becomes invalid, but it stays until I close it.

kocikdav, 2020-11-20:
Hello, you are right that using shortcut to eject should close the notification just like eject button on it. What happens with file after PrusaSlicer exports it, is (in my opinion) beyond what should PrusaSlicer detect. As of timing and fading of notifications, it is matter of testing and adjusting so thank you for your opinion. We will concider it.

kocikdav, 2020-11-20:
Close notification after eject in a0239a75095eaff97cd2260de0363dc9b6f1d72a

closed 2020-11-20 by kocikdav