prusa3d/PrusaSlicer #5195: Feature request:

eracoon, 2020-11-18:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Area5142, 2020-11-22:
This would also enable color change in the infill instead of the perimeter, hiding the transition (perimeter blops).

Jimvanhazendonk, 2020-11-23:
> This would also enable color change in the infill instead of the perimeter, hiding the transition (perimeter blops).

Exactly. I have not seen this function in any other slicer. that would be a massive advantage for me.
So. Manual materials change would be great and maybe assigning different materials to print feature types like infill, perimeter or support.
I'm open for it 👍

fe60, 2021-03-06:
Similar to #2967

TickTock-, 2021-08-30:
Related request:
[Print *all* perimeters before infill](https://github.com/prusa3d/PrusaSlicer/issues/6854)

Bumblebiber, 2023-02-22:
Upvote!

Jimvanhazendonk, 2023-02-22:
yeah, right? I think its an awesome idea. I would enable me to do
colorchanges on exact positions in the print... or other crazy stuff

Bumblebiber, 2023-02-22:
> yeah, right? I think its an awesome idea. I would enable me to do colorchanges on exact positions in the print... or other crazy stuff
> […](#)
> -- Jim van Hazendonk [www.racoon.media](http://www.racoon.media)

Also it would only be ~10 minutes of programming since this function is already there. Small effort, huge impact. It's sad they don't listen to their fans. This request is almost 2.5 years old. :(

Jimvanhazendonk, 2023-02-22:
Agreed! Is there a way to push this forward or make more noise for this.?
like you said... it's a small feature and easy to implement. would make
Prusa slicer that even better for me.

Bumblebiber, 2023-02-22:
Maybe post this in the Prusaslicer Facebook Group... Or in the SuperSlicer Github

Jimvanhazendonk, 2023-02-22:
Good idea! Let's join efforts

Bumblebiber, 2023-02-22:
I'll post in SS Git

Bumblebiber, 2023-02-22:
https://github.com/supermerill/SuperSlicer/discussions/3663

stepan-martinek, 2023-04-10:
I was just about to write the same request. I'm also sending my vote here.

Jimvanhazendonk, 2023-04-10:
Nice! hopefully this feature will be added some time

StryKaizer, 2023-06-22:
Would be cool!  I'm doing some emboss labels inside boxes, this would allow me to change the text part inside the box only to 1 color, without printing the outside box wall in the same textcolor for 1 layer too...

lindelleric, 2024-03-09:
This would be a fantastic feature! Imagine being able to inlay text or logo _inside_ a layer, like a box lid etc. 

it would require however that the logo/text are printed in it’s entirety before pausing for filament change. It would also probably be very difficult to keep these changes when applying different slicer settings. 

Maybe this could be added as a special property to text and svg? Then a user can mark that text/svg to use a different color. And the slicer can make sure that the printer pauses correctly and that it prints the text/svg with the new color. I.e it doesn’t need to be as general as the pausing at a specific layer.

hawwwran, 2024-05-07:
I just got this exact idea, that I want to change the color after perimeters are done to "wipe" the new color into the infill, to prevent imperfections in perimeter. I knew I wasn't the first. It would be nice this task would get more attention.

hawwwran, 2024-05-07:
Like this:
[image]

Trizam, 2024-05-11:
Yeap, just had the same nead/idea!

hawwwran, 2024-10-29:
I’d be absolutely thrilled if the Prusa development team could dedicate some attention to this feature!

Tupson444, 2026-09-13:
still relevant