prusa3d/PrusaSlicer #5196: HTML links when hover text is enough

GWdd, 2020-11-19:
### Version
2.3.0.a2

### Operating system type + version
Win10 20h2

### 3D printer brand / version + firmware version (if known)
Prusa Mk3

### Behavior
Once upon a time when you hovered over settings there was a quick pop-up of text that described the setting. Terse, but useful.

Now, when you hover, you get a useless link that explains nothing.  Worse, if you accidentally click the text area instead of fine tuning the mouse to the check box you open a browser when all you were trying to do is enable ort disable a setting.

This breaks a very useful and standard feature of graphical user interfaces: Hover Text is important.

Please, go back to NORMAL UI behavior.

OR

Make it optional for those that want to use Plicer with a Phone. Don't force something this silly on the rest of us.

Impact123, 2020-11-19:
Try to hover over the value of a setting itself

GWdd, 2020-11-19:
It is still broken as it is.  An acceptable way to deal with adding hot links inside a UI where users control things is to use a right click to access special features, not let a left click open other applications without some intentional user interaction.  This is an application, not a web page.

At the least, if silliness is the teams desired outcome, let users turn off the unwanted 'feature' ...

bobstro, 2020-11-19:
Also a massive security risk, and a time saver for the ransomware criminals since they only have to make modifications to the URLs and distribute a modified version to the unsuspecting.

Opening up a web connection, particularly a one-click operation with no confirmation, is a bad idea.

GWdd, 2020-11-20:
I'm thinking that the underlying purpose of this 'feature' is a good idea, it is the implementation that sucks.  Providing access to more information related to the control in question is always good: but making it much too easy to accidentally invoke another application that pops up in front of (on top) the app you are using is a defect, not a useful feature.

Reading the many attempts people have tried to defeat the way devs have implemented this (other ill-behaved apps that pop up browsers), it makes me think the idea of doing something nefarious was the intent.

bubnikv, 2020-11-20:
> Reading the many attempts people have tried to defeat the way devs have implemented this (other ill-behaved apps that pop up browsers), it makes me think the idea of doing something nefarious was the intent.

chemtrails?

Area5142, 2020-11-20:
I would suggest a "good old help file" to be invoked instead of an insecure URL to content that may be out of sync with the used version of PrusaSlicer.
The content of the help file could be a snapshot of the information on the web pages.

GWdd, 2020-11-20:
> > Reading the many attempts people have tried to defeat the way devs have implemented this (other ill-behaved apps that pop up browsers), it makes me think the idea of doing something nefarious was the intent.
> 
> chemtrails?
 
>bubnikv Opening a help HTML page from a parameter editor is evil? 4 >hours ago

Voj,  when said HTML page is not under direct control of a user, happens unexpectedly and unintentionally, and the landing page can easily contain malware, yes, the concept is evil.  The only applications that do this that I have used are nefarious and evil: mostly by companies or hacker types farming user data for resale to the lowest bidder.  Aka adware.

And if not nefarious, why would Prusa be so upset a user wants to disable the 'feature' ??  After all, some of us pay a lot of money for bandwidth or have a cap on data, a limited number of bytes we can use; having an app use that bandwidth without verification from the user is just bad coding practice.

Sure, having a link to a help page sounds cool, but why not do what major developers do: write a manual.  Or, if you only want to write HTML, make a HELP file.  

As for any help function being an instant action left click, well, so far, that is something new for apps I have used. I think the closest I have seen to this is something like F1 to open a **local** help dialog.

If you want to use a poorly coded app that makes use of this "one click help" to a web page and see how poorly it works in real life on an app that has been around for years, install Quickbooks Desktop Pro and try using it for a while.  They were even more helpful and added a direct link to Internet Explorer.  What Intuit has found is that changing old code is expensive, so they just don't change old code: similar to what Prusa does with Plicer. Old stuff that is bad is still there because changing it is hard work. 
I have low confidence these HTML links will be maintained properly - especially considering the fact Prusa has yet to release a manual of any sort. How many years has Prusa been making their own version of Slic3r now, and promising a manual???  But I digress.

Do the right thing: Add a HOT LINK ENABLE in preferences. At the single line in the code that calls the help browser - just put in a check for the HOT LINKS ENABLED flag. This will only take the team five minutes if this entire scheme was coded properly.

foreachthing, 2020-11-20:
> I would suggest a "good old help file" to be invoked instead of an insecure URL to content that may be out of sync with the used version of PrusaSlicer.

Why would it be out of sync? This help file can be stored and maintained  on github as well. Or, a local copy of the website, maybe?!

GWdd, 2020-11-20:
> Why would it be out of sync? This help file can be stored and maintained on github as well. Or, a local copy of the website, maybe?!

Clearly you have very little experience using apps that have hard links to web sites.

What happens is the app is updated, the web site is not. It is a cost issue. A developer will relegate resources, then when money gets tight the "help" resource is the one that always loses.  Whether the file that is need to stay current is on  github (useless for the average user), a web site (proprietary and generally slow to receive updates), a file on the system that is installed along with the application is always 'correct' for the application. Then, when an update occurs, the help (if it contains updates) can be installed along with the new code.

This way -  a person using older code 2.1.0 has the help appropriate to the app he is using, not some new link totally unrelated to the issue he is seeking help for; or the more likely, help text on a stagnant website that hasn't been updated for a year or three - text more appropriate for code released three years ago.

foreachthing, 2020-11-20:
> Clearly you have very little experience using apps that have hard links to web sites.

Zero to none! :-)
And because I misread! I read that the _help file_ will be out of sync!
To have a website with the help and this many different versions of software out there seems like a bad idea from the get go! Like you just explained! 
What are the guys at [Blender ](https://docs.blender.org/manual/en/latest/)using for their manual? Because you can switch between versions...

I just hope that those links won't go 404 on us...

GWdd, 2020-11-20:
Friends of the IP address help.prusa3d.com resolves to: Instills confidence?  All help will be coming from a third rate no-name web hosting service located somewhere in eastern Europe all while sharing an IP address with other clients.   

```
Pinging prusa3d.net [77.78.104.3] with 32 bytes of data:

  | Domain | View Whois Record | Screenshots
-- | -- | -- | --
1. | klidanko.cz |   |  
**2. | lilyofthedesert.ru |   |**  
3. | smajlici.info |  
```

In a couple years, when you look back at this, I hope you have learned something.

bubnikv, 2020-11-21:
Also the discussion was opened in 
https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-2-3-0-alpha3-is-released/
by @GWdd 

To make everybody happy, we will open a dialog, asking whether you want to open the browser or not, and we will allow you to remember your choice and not ask again. 

This will be implemented in PrusaSlicer 2.3.0-alpha4. Closing.

closed 2020-11-21 by bubnikv