prusa3d/PrusaSlicer #5197: [2.3.0-alpha3] Filaments presets are not imported

Zazou89, 2020-11-19:
### Version
2.3.0-alpha3

### Operating system type + version
Winwdows1 0- 64 bits

### 3D printer brand / version + firmware version (if known)
Artillery Genius

### Behavior
1. Open Wizard Config
2. In Filaments, make choice
3. Finish
4. Presets are missing in Filaments tab

[image]

[image]

rtyr, 2020-11-19:
Expected behavior. These filament profiles are not compatible with your printer. Closing.

closed 2020-11-19 by rtyr

Devilscave, 2020-12-06:
But whit my MK3s i have the same problem, under version 2.3.0 Beta1, I can't add a single one, not even prusament.

rtyr, 2020-12-06:
Seems like different issue. Can't comment on that since there is not enough information.