prusa3d/PrusaSlicer #5198: Poor quality of bottom-facing surfaces atop support

mbrown85209, 2020-11-19:
### Version
2.3.0-alpha3+win64-202011041552

### Operating system type + version
Windows 10 (build 18363), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S, 3.9.1-3518

### Behavior
Problem:
The bottom layer and all orthogonal top-facing layers (as well as sides) produce a nice quality surface finish. But all bottom-facing surfaces (AKA, bridges) that are laid atop supports have a very weak and poor quality surface finish.  And the removal of interface layers is difficult without damaging what infill surface there is.

I will be receiving an MMU upgrade soon, and I expect that the same surface quality issue will exist using a BVOH filament for the Support Material Interface (notwithstanding easier removal).

By all indications, PrusaSlicer is an excellent product; but in order to achieve "Best in Class" status, you really need to analyze your competitors for yourself.  The forums refer to lots of features like this one, that other slicers (specifically, ideaMaker) handle much better.

To reproduce the problem:
Any part with a bridge, sliced with the default, or any other parameters I can think of.

Expected Results:
Create a hardened top surface of the support material interface, then create a bottom-facing surface with the same kind of surface quality (including the Bottom Fill Pattern) as is found on the bottom later.

Actual Results:
Support material interface:
[image]
Overhang perimeter and bridge infill:
[image]
Perimeter and solid infill:
[image]

Is this a new feature request?
Yes

#### Project File (.3MF) where problem occurs
[Fly Wheel test.zip](https://github.com/prusa3d/PrusaSlicer/files/5569760/Fly.Wheel.test.zip)
(change .zip filename extension to .3mf)

mbrown85209, 2020-11-21:
Ok. now I think I see the solution - with a soluble interface layer I can set the Contact Z Distance parameter to 0.0, which omits the crappy Bridge Infill Layer.
But I still think that the Bridge Infill Layer does more harm than good in the absence of soluble material.  Perhaps making a removable layer of the support that LOOKS LIKE the Bridge Infill Layer (omitting the Overhang Perimeter, and extending it beyond the perimeter) would be easier to remove without leaving the final part with such a poor surface finish, as it now has.

bubnikv, 2021-03-18:
To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio. 

[image]

Implemented with ceea9de8b8b1d4b35bdd1180051033680c405ba1 f01f02154c617cc24835475da419ed6322f2ceec

closed 2021-03-18 by bubnikv