prusa3d/PrusaSlicer #5201: Parts come out mirrored

3eggert, 2020-11-20:
### Version
PrusaSlicer 2.3.0-alpha3

### Operating system type + version
Ubuntu 20.04.1 LTS

### 3D printer brand / version + firmware version (if known)
Ender 5 Plus 

### Behavior
* I print out some labels and the writing is mirrored. 
* When I mirror them in the slicer, so that they look mirrored in the slicer view, they come out correctly

#### Project File (.3MF) where problem occurs
The 3mf is the one that comes out right
[Waldau_skylt.zip](https://github.com/prusa3d/PrusaSlicer/files/5574799/Waldau_skylt.zip)

neophyl, 2020-11-20:
Does this happen with other slicers too ?  The most common reason for parts coming out mirrored on new printers is one of the stepper motors wiring being reversed, so has this printer always done this ?

3eggert, 2020-11-20:
I did not check other slicers, I'll try tomorrow. But I would suspect the homing routine to fail if the steppers go in the wrong direction because it would move away from the endstop instead of towards and hitting it.

3eggert, 2020-11-21:
Its also mirrored with the 2.2 version of Prusaslicer. I also checked now on http://gcode.ws/ and it confirms the error is in my printer and not in the slicer. Since the firmware homes correctly I suspect the issue is on all similar creality printers.

I'm real sorry for this wrong alert.

closed 2020-11-21 by 3eggert