prusa3d/PrusaSlicer #5203: Provide hyperlinked graphic examples of the effects of ALL parameters in the Online Help Center

mbrown85209, 2020-11-20:
### Version
2.3.0-alpha3+win64-202011041552

### Operating system type + version
Windows 10 (build 18363), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S, 3.9.1-3518

### Behavior
* _Describe the problem_
Many of the entries in the Online Help Center say the same thing as the tooltip.
Some of them are great. 
Do so for all parameters.

_Is this a new feature request?_
No, just do more of what little has been already done.

#### Project File (.3MF) where problem occurs
N/A

bubnikv, 2020-11-21:
I don't get it. What is your request? To extend the help?

mbrown85209, 2020-11-21:
Yes. For example, click on Top Fill Pattern in the Print Settings/Infill section:

 

 

Samples of fill patterns would be nice, but instead, all you get is this:

 

 

Which tells me nothing more than the tooltip.

bubnikv, 2020-11-21:
Sorry, but pictures are missing.

mbrown85209, 2020-11-21:
Hmm… the email included them, but github didn’t display them.

So I edited the gethub comment, and now it just shows a hyperlink to somewhere it’s stored.

bubnikv, 2020-11-24:
BTW we implemented exactly that: A hyperlink into our knowledge base. You need to click on the label.

See the change log of https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha3
https://user-images.githubusercontent.com/5830947/98116430-8a62c580-1ea8-11eb-89b3-01d7d370d390.png

mbrown85209, 2020-11-24:
The keyword in the request you have missed is “ALL”.  

Only SOME parameters have hyperlinks, and some of THEM only give a brief redundant description.

My request is to provide hyperlinks for ALL parameters, and include images of the effect of changing the parameter value or option.

bubnikv, 2020-11-25:
IMHO this is a request to our content team. I have passed your request to them.

bubnikv, 2020-12-07:
Closing, not much to do with PrusaSlicer team, but with the content team.

closed 2020-12-07 by bubnikv