prusa3d/PrusaSlicer #5205: PrusaSlicer generates multiple superimposed rafts on multi object part.

nasone32, 2020-11-21:
### Version
2.0.3 alpha 3 win64

### Operating system type + version
windows 10 x64

### 3D printer brand / version + firmware version (if known)
not important

### Behavior
after splitting a part in multiple objects, and selecting to generate 1 layer of raft, a raft is generated for every part on the 1st layer, they are superimposed on the same layer causing the tool to extrude multiple times in the same place (thus breaking the print)

* _Steps needed to reproduce the problem_
-Download and open the Gyro air https://www.thingiverse.com/thing:765365
-load the part and select split into objects
-slice using a raft
look at the toolpath in the first layer. it draws multiple passes of dfferent sizes raft.

#### Project File (.3MF) where problem occurs
[Dalpek_Gyro_Air_Big_v6.zip](https://github.com/prusa3d/PrusaSlicer/files/5577680/Dalpek_Gyro_Air_Big_v6.zip)

bubnikv, 2020-11-21:
PrusaSlicer slices each object separately. You should not have split the object.

closed 2020-11-21 by bubnikv

nasone32, 2020-11-21:
I needed to split the object because I needed to resize the different parts at slightly different scales each One.

This is definitely a bug! Because if you have more parts on the plater it can potentially break like this again.

bubnikv, 2020-11-21:
You can resize the parts one by one even inside a single object.
You can also split the object, then resize the parts, then merge the
objects by selecting them all, opening the pop-up menu by right clicking
and selecting "merge".

Leooooooooooooooooooooooooooooo, 2024-06-20:
For those who join this years later, here is the trick I have used :
- Export build plate as STL with all the objects
- Import it back as one single STL
- Activate raft
Now I have a single raft for my multiple pieces.