prusa3d/PrusaSlicer #5206: Solid Infill every X layers are not solid

mbrown85209, 2020-11-21:
### Version
2.3.0-alpha3+win64-202011041552

### Operating system type + version
Windows 10 (build 18363), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S, 3.9.1-3518

### Behavior
 _Describe the problem_
Specifying a Solid Infill every X Layers value inserts what appears to have the density of the Bottom Infill Layer, which produces a weakness in the part, instead of added strength.

_Expected Results_
Make the added layer truly solid.

* _Actual Results_
[image]

_Is this a new feature request?_
no

bubnikv, 2020-11-24:
Dense infills over sparse infills are extruded using bridging flow.

bubnikv, 2020-12-07:
Works as intended.

closed 2020-12-07 by bubnikv