prusa3d/PrusaSlicer #5207: auto arrange and arrange selection issues

screwyluie, 2020-11-21:
### Version
2.2 alpha 4

### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
prusa mk2.5s

### Behavior
add a benchy
right click fill bed with instances (that's a different issue because that's not right either)
then select all
hit the arrange button.

now select all
arrange selection.

these do not match as they should
#### Project File (.3MF) where problem occurs

[files.zip](https://github.com/prusa3d/PrusaSlicer/files/5578093/files.zip)

screwyluie, 2020-11-21:
not sure if this should be a different issue but I think it may be related on the backend... but allow rotation leaves a weird gap as well when you auto arrange.

[image]

Area5142, 2020-11-22:
On one of the images I see a purge block - maybe the auto arranger is making room for that on the build plate.
I have strange arrangements, using MMU profile with purge block as reported in #5216.

paulamath, 2020-11-22:
If you rotate the piece 45 ° and then click on auto arrange to center on the table, the piece centers but undoes the rotation.

commit f1acde80 (2020-11-23): Fix broken wipe tower arrangement

fixes #5207
fixes #5216
closed 2020-11-23 by tamasmeszaros