prusa3d/PrusaSlicer #5208: [2.3.0-alpha4] Crash - several models cause crash when slicing

tg73, 2020-11-21:
### Version
2.3.0-alpha4 Win64

### Operating system type + version
Win10 x64

### Behavior
This is happening with several, but not all, models for me. The simplest reproduction case I can find is attached.

Attempt to slice with 0.15 layer height, PS exits without warning. This does not happen at 0.20 layer height. I suspect that there's some odd interaction between layer height and the specific model. The attached model slices fine with the same settings in 2.2.0.

#### Project File (.3MF) where problem occurs
[files.zip](https://github.com/prusa3d/PrusaSlicer/files/5578125/files.zip)

bubnikv, 2020-11-21:
Yeah, it crashes in the new algorithm connecting infill lines along the
inner most perimeters.

ondras, 2020-11-22:
I have another crash in alpha4, not sure if related.

Load https://www.thingiverse.com/thing:2755765, **set scale to 50%**, slice -> crash. Crashes with both 0.15 and 0.20.

MarkWheadon, 2020-11-22:
I'm ticking perhaps the same bug – crashes as soon as this model is placed sensibly on the bed and then sliced at 0.05mm ultradetail, Original Prusa Mini.

[3-1x-frontLensPillar-converted.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/5579967/3-1x-frontLensPillar-converted.stl.zip)

bubnikv, 2020-11-23:
It crashes at a special infill configuration, where just a single infill line is drawn from an outer contour to a hole in an infill island. We will likely release another alpha (or beta?) in couple of days just due to this bug.

foreachthing, 2020-11-23:
Huh...?!?!
Cannot confirm any of your crashes on my end?!

[image]

sej7278, 2020-11-23:
@ondras i have mentioned the scaling crash in #5229 i think its two very related issues - infill crossing a boundary is the main issue, which is triggered (or at least happens more often) by scaling up or down or even using a finer layer height or changing infill type.

the segfault i'm seeing is in `mark_boundary_segments_touching_infill()`

bubnikv, 2020-11-23:
@ondras Please post a 3MF, so we can retest.

ondras, 2020-11-23:
> @ondras Please post a 3MF, so we can retest.

I do not have any - only the STL linked in my previous comment. Is that not enough?

bubnikv, 2020-11-23:
For reference, these are the 3mfs created from the two other models mentioned. I was able to reproduce crash with them.

[3-1x-frontLensPillar-converted.zip](https://github.com/prusa3d/PrusaSlicer/files/5584434/3-1x-frontLensPillar-converted.zip)
[ella_oc_a015R.zip](https://github.com/prusa3d/PrusaSlicer/files/5584435/ella_oc_a015R.zip)

A fix is in the workings.

bubnikv, 2020-11-24:
Fixed with 62bdc192d87a9a1835e6ff6ff10b5a0893279c98

closed 2020-11-24 by bubnikv

MasterCATZ, 2021-01-31:
PrusaSlicer-2.3.0+linux-x64-202101111322.AppImage

segfault 
Ubuntu 20.04 Kernel 5.10 

[slicecrash.log](https://github.com/prusa3d/PrusaSlicer/files/5899081/slicecrash.log)

[snorkle.zip](https://github.com/prusa3d/PrusaSlicer/files/5899082/snorkle.zip)