prusa3d/PrusaSlicer #5209: Bad support generation with default settings in Alpha 3

thetrebor, 2020-11-21:
### Version
2.3.0-alpha3+

### Operating system type + version
OSx 10.15.7

### 3D printer brand / version + firmware version (if known)
Stock prusa mk3s

### Behavior
Use this model: https://www.thingiverse.com/thing:4128842
Use default profiles.  Tried 0.2 and 0.1
[image]

Support blocker does *not* work unless it is in contact with the model.
Repaired the model using ADMesh and did not change outcome
Changed support overhand angle and did not change outcome

Model appears OK to me.  Didn't see any loose geometry.

neophyl, 2020-11-22:
Support enforcers and blockers do not work unless they intersect with the model, thats the way they are supposed to work.  They aren't keep out areas in the case of blockers.

I downloaded it and used the new support painting function, but I just used autoset by angle to 35 degrees which is sufficient for this model.  
This was in PS alpha 4 as that's what I have installed.

The supports look reasonable to me, although as they are printing on top of parts of the model I would expect them to be difficult to remove on that side from experience.  

[image]

I also did try it with a slab support enforcer and the pattern of supports was similar but did grow outward slightly from the painting as its also supporting the part more on the less steep angles too as the slab is less discriminate.

[PS_alpha4_WyvernPhoneStand.zip](https://github.com/prusa3d/PrusaSlicer/files/5578716/PS_alpha4_WyvernPhoneStand.zip)

thetrebor, 2020-11-22:
thank you for trying to reproduce.  i couldn’t figure out what in the model was making prusa think the dead area in front of the head needed support.  i suspect there’s a bad detection happening somewhere which results in an infinity and the result is presumption that supports are needed to the edge of the bounding box in that direction.  given that, it could be graphics card specific or driver specific.  i don’t know that i’m qualified to diagnose that though.

bubnikv, 2020-11-24:
@thetrebor Would you please attach a 3MF so we can reproduce?

thetrebor, 2020-11-24:
[WyvernPhoneStand.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5591942/WyvernPhoneStand.3mf.zip)
[image]

Fresh install, default profiles, added default "supports everywhere"

bubnikv, 2020-11-25:
Thanks, I can reproduce the issue.

There is a numeric issue in one of the algorithms in succession to the
Clipper offsetting library creating tiny self intersections in an offsetted
contour. I have not seen this kind of issues being reported for along time
though.

thetrebor, 2020-11-25:
that seems reasonable.  I saw that some of the cusps on the model were being supported despite having no overhangs whatsoever.  it appeared to be "self intersecting" "non manifold" style problems so my first thought was to repair the model.  Nice to know I'm not crazy.  Good luck tracking it down!

bubnikv, 2020-11-25:
For my reference, the issue is 
void EdgeGrid::Grid::calculate_sdf()

HiLivin, 2020-11-28:
2.2.0+win64
Excessive support is generated quite far from the object's outline. The object was imported from STL file and rotated.
**Interesting thing is that saving the project in PrusaSlicer as 3mf file, re-opening it and re-slicing doesn't generate the excessive support.**
 
[image]
[Bracket_without_inserts.zip](https://github.com/prusa3d/PrusaSlicer/files/5611121/Bracket_without_inserts.zip)
[Bracket_without_inserts-3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5611125/Bracket_without_inserts-3mf.zip)

commit fb7b9950 (2021-02-19): Improvements in FDM support generator:
Projection into a grid has been reworked to use the AGG rasterizer.
This fixes #5209 and #6067.
Also the raster is now being oversampled by maximum 8x8 samples
and the supports are only allowed to expand inside the cell.
This significantly reduces leakage of supports through object walls,
which fixes #5054.
closed 2021-02-19 by bubnikv