prusa3d/PrusaSlicer #5210: Crash when showing Shell in GCode-Preview with specific model

dartrax, 2020-11-21:
### Version
2.3.0-alpha3+win64, 2.3.0-alpha4+win64

### Operating system type + version
PrusaSlicer
Version:   2.3.0-alpha4+win64
Build:     PrusaSlicer-2.3.0-alpha4+win64-202011211615

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 17,075MB
OpenGL installation
GL version:   4.6.0 NVIDIA 425.45
Vendor:       NVIDIA Corporation
Renderer:     GeForce MX150/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
[... 319 lines cut ...]
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### Behavior
* _Describe the problem_
Crash in GCode-Preview when View=FeatureType in the moment of enabling Show Shells with "Lichtleiter.stl"
I tried another stl, that worked.
If I try "Lichtleiter.stl" together with that another stl, it still works.
If I try "Lichtleiter.stl" after I deleted the other stl, it still works.
So it only crashes when the PS instance is fresh in the means of not having displayed Shells before.

* _Steps needed to reproduce the problem_
Start PS, load Lichtleiter.stl or Lichtleiter.3mf, slice, switch to GCode Preview, enable Shells

#### Project File (.3MF) where problem occurs

[Lichtleiter.zip](https://github.com/prusa3d/PrusaSlicer/files/5578191/Lichtleiter.zip)

bubnikv, 2020-11-21:
Would you please try with alpha4? I cannot reproduce your issue with alpha4.

dartrax, 2020-11-21:
This happens with 2.3.0-alpha3+win64 and 2.3.0-alpha4+win64
I've made a little video about that, see below.
Are there any log files for that?

[IMG_5013.zip](https://github.com/prusa3d/PrusaSlicer/files/5578286/IMG_5013.zip)

bubnikv, 2020-11-24:
Sorry, I cannot reproduce it.
@enricoturri1966 @FidelCapo could you please try to reproduce it?

dartrax, 2020-11-24:
I've tried on a different computer, the HP Z440 Workstation I use at work, and it is also crashing, I could reproduce.

Version: 2.3.0-alpha3+win64
Build: PrusaSlicer-2.3.0-alpha3+win64-202011041552
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 34,272MB

PrusaSlicer
Version:   2.3.0-alpha3+win64
Build:     PrusaSlicer-2.3.0-alpha3+win64-202011041552

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 34,272MB
OpenGL installation
GL version:   4.6.0 NVIDIA 390.77
Vendor:       NVIDIA Corporation
Renderer:     Quadro M2000/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
[... 303 lines cut ...]
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

Eigen vectorization supported: SSE, SSE2

FidelCapo, 2020-11-25:
I cannot reproduce this issue. I tested it on all platforms (MacOS, Win, Linux)

dartrax, 2020-12-07:
Still there in 2.3.0-beta2+win64 at least for me. I'm using 2 monitors, same resolution, if that makes a difference...

bubnikv, 2020-12-07:
Anybody else would please try to reproduce this issue?

bubnikv, 2020-12-08:
tested on another 4 computers at PrusaResearch, works fine.

bubnikv, 2020-12-08:
@YuSanka tested with alpha4.

From your movie compared with the file you provided:

The preset combo boxes differ:
[image]

Also the timestamps differ:
[image]

Is it possible that you attached a wrong file?

dartrax, 2020-12-08:
Thanks @bubnikv for looking into it that eagle-eyed ;-)
I think I can rule out that the wrong file is attached. I tried it once again, downloading the file here from github. I've attached a video.
Both PCs I have tested with have an nvidia graphics card. In the past, I had another issure #1909 that was only reproducible on those machines. In that case, "OpenGL legacy immediate mode (or its implementation in [my] driver) was the culprit."

[2020-12-08 16.00.01.zip](https://github.com/prusa3d/PrusaSlicer/files/5660120/2020-12-08.16.00.01.zip)

I think we'll have to wait if that issue is more widespread or related to my specific config somehow. I can live with that, don't spend too much time if it's not a widespread issue.

Impact123, 2020-12-08:
I can reproduce this on my main pc with a nvidia gpu but not in hyper-v nor a second pc with only intel hd graphics. Disabling/enabling `Background processing` and restarting seems to "fix" this but i only tested briefly

bubnikv, 2021-09-22:
Is it still an issue with PrusaSlicer 2.4.0-alpha1?

dartrax, 2021-09-25:
In 2.4.0-alpha2 it does not crash, while in 2.4.0-alpha1 it did crash.
So this should be fixed, closing.

closed 2021-09-25 by dartrax