prusa3d/PrusaSlicer #5211: Sort loaded STLs by Name or Object Printability by clicking table headers

J-Hutt, 2020-11-21:
### Version
2.3.0-alpha4

### Operating system type + version
Win10-64

### 3D printer brand / version + firmware version (if known)
PrusaMK3s

_Is this a new feature request?_ Yes
Allow loaded STL files to be sorted by clicking on the Name header (alphabetically) or Object Printability Property header (on/off) in the sidebar table.
This would allow loading of large sets of STLs, and for them to to be better sorted, currently printing items grouped/identified.

Currently clicking on table header does nothing.

neophyl, 2020-11-22:
The order of the list on the right is used to determine print order, as such its not just a convenience of viewing thing, it actually is important.  If you sorted the list lie you suggest you would be changing the print order too.

J-Hutt, 2020-11-22:
So a re-sort would trigger the need to re-slice? Not really any different to deleting an STL and then re-adding it, is it?

neophyl, 2020-11-22:
No I think you misunderstand my point, the order of that list has a specific function within the software,  its not just something for easy ordering like a file sort.  You can reorder the list by dragging them around.  It will use the new order the next time its sliced but unless you have background processing on (which I don't have enabled as it drives me crazy) then it it shouldn't re-slice when you change the order.  If it is on then yes changing the order does trigger a reslice.

Just tried it with alpha4 and if I add multiple stl files at the same time it does add them into the list in alphabetical order.  After that it adds them in the order they are added.  So that all makes sense to me.  If I wanted a lot of parts loaded in alphabetical order I would just add them all in at the same time.

J-Hutt, 2020-11-22:
Understood, thanks for the explanation

rtyr, 2021-02-25:
I believe we can close this. Thanks @neophyl

closed 2021-02-25 by rtyr