prusa3d/PrusaSlicer #5212: Feature Request: Gap Fill - Option to delay first layer gap fill until after perimeters are laid

JasonGuthrie, 2020-11-21:
Currently on Prusa Slicer, when printing objects with text (or fine details) on the bottom surface, there is an issue where 'Gap Fill' is triggered as the very first operation the printer performs on the first layer. This causes an issue, as the little blobs are laid down very quickly but with nothing to stick to and then detach, getting stuck to places in the print where they shouldn't be.

If an option could be put in to set 'gap fill' to be the last operation (or to commence after the perimeters have been laid down) it would alleviate this issue and help to improve overall first layer quality.

### Version
_Version of PrusaSlicer used goes here_
2.3

_Is this a new feature request?_
Yes

Area5142, 2020-11-22:
I have this problem too when printing text directly to the build plate (first layer).
Tiny small dots/extrusions is placed first and some of them breaks loose and attaches to the nozzle (PETG).
This is especially annoying when using MMU and multi-color text as the colors gets mixed and the appearance and clarity of the text is spoiled.

I think this is a bug as there is no reason for gap fills to be printed before perimeters.

JasonGuthrie, 2020-11-28:
> I have this problem too when printing text directly to the build plate (first layer).
> Tiny small dots/extrusions is placed first and some of them breaks loose and attaches to the nozzle (PETG).
> This [...]

Yes I agree, the problem seems most pronounced with PETG (although I have to admit my experience with other filaments is not that expansive). 

I also agree that this is a bug more than a feature request now that I think of it. A change to the operation order on the first layer should solve this issue.

morphias2004, 2020-12-03:
Yep. Reported this issue too about 6 months ago.

Makes printing text into control panels a nightmare and results in many failed first layer prints.

PLEASE PLEASE PLEASE add this feature @josefprusa

closed 2026-09-29 by prusaslicer-bot[bot]