prusa3d/PrusaSlicer #5213: [Feature Request]  A feature in GCODE viewer that lets you view, well, the actual GCODE

MR2C280, 2020-11-21:
It would be helpful to be able to view (and edit?) the GCODE from within the GCODE viewer.  It should have a Search (and Replace?)   function.  It should only show the GCODE lines indicated by the layer and toolpath sliders.

foreachthing, 2020-11-22:
I like how CrafWare does the preview. It shows an overlay with the current gcode. 
But search/replace is a case for notepad(++).

bubnikv, 2020-11-24:
> I like how CrafWare does the preview. It shows an overlay with the current gcode.

We do like it too.

rtyr, 2021-04-27:
Implemented on master. It will included in PS2.4.0. Closing.

[image]

closed 2021-04-27 by rtyr

foreachthing, 2021-04-28:
Looks good, but it just starts where it wants. I can't select my gcode from or bevore the homing procedure.
I'd like to see how the nozzle moves while heating and so on ...

rtyr, 2021-04-28:
It should be better if you enable travel moves in the preview.

foreachthing, 2021-04-28:
> It should be better if you enable travel moves in the preview.

Awesome!!! 👍🏼