prusa3d/PrusaSlicer #5214: Feature Request: Archimedean Chords - Option to define direction of travel (Inside-Out or Outside-In)

JasonGuthrie, 2020-11-21:
I'd like to request the option to define the direction of travel on the top/bottom infill (with specific relation to Archimedean Chords) so that it is possible to define if we want the infill to be laid from center out or from outside in.

Currently the when using Archimedean Chords, the only behavior is for the toolpath to start from the outside and work it's way into the middle. The issue with this mainly apparent on the top layer, whereby excess plastic is forced towards the center more and more with each pass. This then causes a blob to form in the innermost passes, ruining the surface finish.

If the toolpath could start from the center and work it's way out, the excess plastic would be evened out between each pass and I suspect that by the time it reaches the outer edge it will even out and become perfectly flat.

# Version
_Version of PrusaSlicer used goes here_
2.3

_Is this a new feature request?_
Yes

LordMelkor, 2020-11-23:
I second this request--I actually just logged into request this exact issue.

@JasonGuthrie isn't exactly right on the behavior of the current implementation, since it appears PrusaSlicer alternates between outside in and inside out (although it's unclear under which conditions it chooses inside-out)

Inside out looks really fantastic, however there does not seem to be any way to configure which configuration the slicer chooses.

As mentioned above, the outside-in completely distorts/ruins the center of the spiral. If not an option to configure the toolpath, then an option to never let it do outside-in may also be reasonable.

JasonGuthrie, 2020-11-23:
@LordMelkor thank you for the correction, I wasn't 100% sure if it was always one way or the other. Each time i've tried to use it, it has gone outside in and caused these issues (so i've not persisted with it that much, but I would really love the option to have consistent concentric patterns on all object top surfaces!).

LordMelkor, 2020-11-25:
[image]

JasonGuthrie, 2020-11-28:
> Top row is inside-out, bottom row is outside-in
> 
> [image]

Yep this is the exact issue (actually for me it's been much worse than this, when using PETG at least). I have also had the same issue with the infill not quite filling out the outer perimeter with this pattern too (when the slicer decides to do 'outside-in').

sg000, 2020-12-23:
Similar issue with Concentric fill pattern. In my case, I need it to fill outside to in rather than inside to out. Having a 3-way selection for (slicer optimized [today's default], inside-out, outside-in) for both infill options would resolve this quite easily for all of us.

bubnikv, 2021-01-05:
Adding a three-way switch for the concentric infill would be a small step towards making the software an unmaintainable mess for us developers and a UI nightmare to the users. One of our employees bugs me for a long time to make the direction always outside-in. 

What is the thickness of these prints? A single layer? Why are there gaps and bulges in your print? I have to reprint myself.

There may actually be an issue with the infill-perimeter overlap, which may be counter-productive for the concentric infill.

bubnikv, 2021-01-05:
@JasonGuthrie Archimedean Chords or Concentric?

sg000, 2021-01-05:
> Adding a three-way switch for the concentric infill would be a small step towards making the software an unmaintainable mess for us developers and a UI nightmare to the users. One of our employees [...]

I can only speak for myself (I'm not the original poster). Its the Concentric for me. And I agree with your employee. Thus far I've only encountered where I would want it outside to in because you run the perimeter then work towards the inner.
My issues always occur on the first layer. And my prints are rectangular in nature, not circular like the OP.

Isn't the REAL question is why is the direction (out->in vs in->out) changing not only between prints but also from release to release? Why does it choose one way or the other?

JasonGuthrie, 2021-01-05:
> @JasonGuthrie Archimedean Chords or Concentric?

Chords for me.

Concentric causes zero issues for me as the infill's toolpath is not one continuous sweep like it is for Chords (on objects with a featureless and circular base/top), and therefore does not cause a centripetal build-up of excess material.

>What is the thickness of these prints? A single layer? Why are there gaps and bulges in your print? I have to reprint myself.
>There may actually be an issue with the infill-perimeter overlap, which may be counter-productive for the concentric infill.

This issue has presented itself both on first layer and on top layer. I have not experimented with the setting much further due to the large proportion of scrap parts I was getting.

SuperSlicer has the option to 'fill' these infill patterns, which alleviates the issue of can gaps caused by the infill pattern (although I am not too sure how relevant that is to this particular issue of infill direction, so I might be missing your point and apologies if I am). 

Since the introduction of the step-by-step g-code viewer in the Sliced View, I can at least now see how the Chords are going to behave and I can omit the infill if it looks like it is going to cause a build-up at the center.

sg000, 2021-01-05:
> I think the other user is confusing the issue by bringing up concentric. That is a separate issue.
My apologies. Totally understand. I was just jumping on the direction portion of it which seems suspiciously similar to my problem with concentric. I'll stay out from this point.

bubnikv, 2021-01-06:
Implementing a outside-in or inside-out path planning for the Archimedean Chords infill sounds significantly more involved than the fix for the concentric infill.

[image]

Indeed, what you are really asking for is a kind of Montonous infill for the corners that I marked with green color before jumping into the continuous spiral.

[image]

JasonGuthrie, 2021-01-10:
@bubnikv 

I guess so, to be honest I never considered using this infill for anything that wasn't circular, due to it not fitting the shape particularly well. 

Regarding the corners on your example, I am not overly fussed which way the toolpath runs outside of the center spiral as it is no longer a continuous sweep. 
Is it not possible to define the center spiral's toolpath in isolation to the rest of the infill i.e. if the first travel move is greater than 0mm, switch to the shortest infill path? 
This is just a 'thinking out loud' suggestion and I apologise that I cannot be of more practical use with the coding suggestions, unfortunately it is far beyond my area of expertise. 

It is honestly not a deal breaker when using PS, I just ignore Chords for the most part unless the slicer chooses to run the toolpath 'inside-out'. If this issue is too much work to solve in comparison to the issue it solves, please feel free to close the issue. 

Thanks for all your hard work!

LordMelkor, 2021-01-10:
The outside-in or inside out toolpathing should only apply the the portion that forms a continuous

> Implementing a outside-in or inside-out path planning for the Archimedean Chords infill sounds significantly more involved than the fix for the concentric infill.
>
> [image]
>
> Indeed, what you are [...]

To address the printing issue identified, the outside-in/inside-out logic would only need to apply to the portion of the tool pathing that forms a continuous line down to the center point. If there are any discontinuities in the chord path, it can retain whatever behavior it has today.

The problem arises from the _continuous_ sweeping motion that spirals inward to a point.

bobjackman, 2023-08-24:
Even if monotonic planning were too difficult, it would still be a huge improvement if it wouldn't jump around. It seems pretty random where some sections it'll print a couple lines, then skip ahead and do a few more, then come back and fill in between, then skip WAY ahead, then come back. It's all over the place, currently.

bcegatien, 2024-01-11:
When using Archimedean Chords, I realize that the lines are not always printed in order. Every time a line is skipped and printed out of order, the finish is uneven. I did my best to put side by side a picture of a test print and the G-Code as showed in PrusaSlicer. "Ordered" Archimedean Chords would solve this.
[image]

zandr, 2024-02-23:
I found this issue looking for a switch to control direction because I'm experimenting with the "printing glass" methods but for round objects, and on one of my tests it's actually alternating direction layer-by-layer. I don't much care which direction it goes as long as it's consistent. (This is on 2.7.1)

benjaminhale, 2024-05-14:
I have found that Wall Generator: Classic will make Archimedean Chords go from inside to out on a circular top surface, while Wall Generator: Arachne will make the order inconsistent, with some (in the same slice) be inside-to-out while others are outside-to-in.

Mqrius, 2024-10-05:
I've found that many things will influence the direction. Since we can't choose, at least we can play around until it does what we want. The most convenient ones for me have been:
* Rotating the model some amount can flip the direction
* Changing the line with slightly can flip the direction

I'm not familiar with the underlying code, but as I understand it the top and bottom surfaces are handled by the infill functions. The infill functions simply make line segments, without defining what the start and end are. Some other part of the code presumably decides the best order and direction to draw the lines. Somehow monotonic line manages to define the start and end of line segments though, so there seems to be some pre-existing functionality to do something similar to what we'd want..

benjaminhale, 2024-10-17:
I messed around with these two points that Mqrius made and found something interesting. It appears that the direction is linked to proximity to the wall's seam. If the tail of the spiral (the outermost part where the nozzle begins/ends the extrusion) is too close to the wall's seam, it will go outside-in. If the tail is further away, it will flip to inside-out.

zjohnson16, 2025-01-13:
+1 I don't see any positives in the random jumping around while printing everything besides the inner spiral for the chords. I just have a simple rectangle to print, which seems like it could be split into three regions that could each be printed monotonically.

julian-weinert, 2025-01-28:
@bubnikv 
> Implementing a outside-in or inside-out path planning for the Archimedean Chords infill sounds significantly more involved than the fix for the concentric infill.

It would behave very differently with surfaces containing additional geometry though.

[image]

Whereas you actually would want a result like this

[image]

Here is a preview of the jumping around. This happens in all quadrants and messes with the surface quality

[image]

ilovestuffforreal, 2025-02-13:
I too am interested in this issue. I think having a control on this would be so awesome, especially for bicolored filament. As you can see here, the center of my part looks amazing, yet the rest of it is quite random and messy. 

[image]

iamkim17, 2025-07-22:
Is there a fix for this now? I'm also having trouble with skipping lines using Archimedean Chords as the top layer

louispires, 2026-01-30:
2026 and this is still an issue - The inconsistent lines make this nearly unusable...

benjaminhale, 2026-09-02:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

I'm definitely still interested in seeing this resolved.

JasonGuthrie, 2026-09-07:
@benjaminhale same here, would be a nice option to have in PrusaSlicer 3.0

@bubnikv is this option an easier win with 3.0's new architecture?