prusa3d/PrusaSlicer #5216: Purge block doesn't move when filling bed with instances

tiffyslemonadestand, 2020-11-22:
### Version
2.3.0-alpha4

### Operating system type + version
Windows 10 Home 64 Bit

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Possibly a feature request, purge block is not moving when filling the bed with instances.

[image]

[image]

Area5142, 2020-11-22:
I have seen this behavior in alpha3 as well. It is like the auto arrange is making room for the purge block in the middle of the build plate instead of the actual purge block placement.

Trying to place parts in the middle and auto arrange moves them away from the middle and some parts can be placed into the purge block as your last picture shows.

tiffyslemonadestand, 2020-11-23:
Sorry if I wasn't clear, but this is for the new "Fill Bed" feature.

commit f1acde80 (2020-11-23): Fix broken wipe tower arrangement

fixes #5207
fixes #5216
closed 2020-11-23 by tamasmeszaros