prusa3d/PrusaSlicer #5217: Arrange Options dialogue improvement request

arcady-genkin, 2020-11-22:
### Version
PrusaSlicer 2.3.0-alpha4+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Behavior
In the new "Arrange options" dialogue I can only set the gap size by pulling the slider left and right.
This is frustrating because it's impossible to specify an exact number, for example if I want to set it to 8, I always end up with the likes of 8.407 or 7.965. I would like to be able to type in "8".

Additionally, I think it would make sense for the slider to be movable with the mouse wheel, rather then clicking and dragging, but this is less important.

tamasmeszaros, 2020-11-25:
Hello! Thanks for the feedback. I see the problem, we will do something about it. In the meantime you can do CTRL+left click and the slider should switch to a text field and you can specify any value.

timur-tabi, 2020-12-06:
Ctrl-left click doesn't work on Mac.

bubnikv, 2020-12-07:
> Ctrl-left should be Command-left click on Mac.
fixed with 8bf55273c3ae0513953153ce09959f85616f7af4

closed 2020-12-07 by bubnikv