prusa3d/PrusaSlicer #5218: Wipe tower printing

MaximSharkov, 2020-11-22:
### Version
PrusaSlicer 2.2.0

### Operating system type + version
Windows7 x64

### 3D printer brand / version + firmware version (if known)
home made H-bot (Marlin2)

### Behavior
wipe tower printing must be associated with any tool change, not with multimaterual model(s)

rtyr, 2020-11-22:
I am afraid brief description like this is not enough. Please provide more detailed description, preferably specific case with 3MF project file.

bubnikv, 2020-11-23:
The strategy PrusaSlicer chooses for printing supports with multi-material
print is that the break-away support is printed with the last active
extruder on the layer. Only if you select printing with soluble interfaces
will the extruder loaded with soluble material wiped into the wipe tower.

MaximSharkov, 2020-11-24:
23.11.2020, 11:34, "Vojtěch Bubník" <notifications@github.com>: The strategy PrusaSlicer chooses for printing supports with multi-materialprint is that the break-away support is printed with the last activeextruder on the layer. Only if you select printing with soluble interfaceswill the extruder loaded with soluble material wiped into the wipe tower.  OK. Material changed to soluble, nozzle wiped, support printed. Next, material changed back to main, nozzle NOT wiped, model body  printed with mix of materials while nozzle not wiped from soluble. Single nozzle printers must wipe nozzle on each tool change regardless using soluble or not.  Remark: select for support soluble material (the "soluble" option checked) does not print wipe tower. Wipe tower printed only for multimaterial model body.

bubnikv, 2020-11-24:
Look at the Prusa3D MK3S MMU2 profiles for soluble filaments.

rtyr, 2020-11-28:
You have to set `contact Z distance` to 0.

[image]

bubnikv, 2020-11-30:
Please close the issue then.

closed 2020-11-30 by MaximSharkov