prusa3d/PrusaSlicer #5220: Model not showing up as solid in PrusaSlicer 2.3 alpha 4 but prints ok in 2.2

stillpointx, 2020-11-22:
### Version
2.3.0-alpha4+linux-x64

### Operating system type + version
Ubuntu 64bit 20.04

### Behavior
PrusaSlicer 2.3.0 alpha 4 not seeing object as solid but Prusaslicer 2.2 does

**Prusaslicer 2.2**
[image]

**Prusaslicer 2.3 alpha 4 doesn't work**
[image]

Attached 3mf and stl file for both

[prusa issues with 2.3 alpha 4 files.zip](https://github.com/prusa3d/PrusaSlicer/files/5579138/prusa.issues.with.2.3.alpha.4.files.zip)

rtyr, 2020-11-22:
The missing walls are too thin to be printed with default extrusion width. The different result with the PS2.2.0 is due to the fact that you have the "Detect thin walls" option enabled in `prusaslicer 2.2.0-esd_wrist_snaps-Socket_bkup.3mf`.

Can we close the issue?

stillpointx, 2020-11-22:
Yes Thanks

closed 2020-11-22 by stillpointx