prusa3d/PrusaSlicer #5222: Weird dependency of travel speed on printing time

espr14, 2020-11-22:
### Version
2.3.0 alpha3 win64

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
MINI

### Behavior
Slice a model with default setting and then with travel speed set to e.g. 9999. You'll see that the faster travel speed will cause much longer printing time. The more you set the longer is the printing time. Moreover, it does not take machine limit into account.

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
[MINI-y-plate-front.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5579184/MINI-y-plate-front.3mf.zip)

bubnikv, 2020-11-24:
Můžete být trochu více konkrétnější, pane kolego? Můžete prosím dodat tabulku 3MF versus čas odhadu tisku? Takhle netušíme, co čem se bavíme. Mnohokrát děkuji.

espr14, 2020-11-25:
Stále stejný projekt, jen se mění travel_speed:

printing time @ travel_speed
7h32m @ 50 mm/s
7h18m @ 100
7h16m @ 150
7h16m @ 200
7h18m @ 250
7h19m @ 300
7h28m @ 500
7h45m @ 1000
9h11m @ 9999

S rostoucí rychlostí má čas klesat. Přesněji má to být nerostoucí funkce.

closed 2026-09-29 by prusaslicer-bot[bot]