prusa3d/PrusaSlicer #5224: Nelze slicovat objekt

joka-cz, 2020-11-22:
### Version
PrusaSlicer
Version:   2.3.0-alpha4+win64
Build:     PrusaSlicer-2.3.0-alpha4+win64-202011211615

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 17,099MB
OpenGL installation
GL version:   4.3.0 - Build 10.18.15.4274
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 4600
GLSL version: 4.30 - Build 10.18.15.4274

### Behavior
Při slicování přiloženého projektu PrusaSlicer spadne, aniž by se objevilo nějaké chybové hlášení.

Cca v květnu minulého roku jsem objekt slicoval a vytiskl, model je v pořádku.

#### Project File (.3MF) where problem occurs
[Krytka stroje.zip](https://github.com/prusa3d/PrusaSlicer/files/5579265/Krytka.stroje.zip)

rtyr, 2020-11-22:
Souvisí s https://github.com/prusa3d/PrusaSlicer/issues/5208.

joka-cz, 2020-11-22:
děkuji za informaci, máte reportováno, můj příspěvek zavírám.

closed 2020-11-22 by joka-cz