prusa3d/PrusaSlicer #5226: PrusaSlicer-2.3.0-alpha4+-202011211604.dmg crashes when loading stl

migimli, 2020-11-22:
### Version
PrusaSlicer-2.3.0-alpha4+-202011211604.dmg

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
macOS Catalina 10.15.7_

### 3D printer brand / version + firmware version (if known)
Prusa MK3S 3.9.1 stock kit

### Behavior
opening the image
starting the alpha-slicer
loading benchy dock file.
click slice: crash of app
had this first with autoslicing on: disabling autoslicing makes the crash wait until slicing is done

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
NO

[alpha-crash.zip](https://github.com/prusa3d/PrusaSlicer/files/5579302/alpha-crash.zip)

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-11-22:
Related to https://github.com/prusa3d/PrusaSlicer/issues/5208.

migimli, 2020-11-22:
not a windows thing then :)

bubnikv, 2020-11-24:
Fixed with 62bdc192d87a9a1835e6ff6ff10b5a0893279c98

closed 2020-11-24 by bubnikv