prusa3d/PrusaSlicer #5229: PrusaSlicer-2.3.0-alpha4+-20201121160 crashes during slicing due to a segmentation fault

jaapp, 2020-11-22:
### Version
PrusaSlicer PrusaSlicer-2.3.0-alpha4+-202011211604

### Operating system type + version
MaxOS Mojave 10.14.6 (18G4032)

### 3D printer brand / version + firmware version (if known)
Prusa MK3 MMu2S stock

### Behavior
* Start Prusaslicer from .dmg image
* Create default Prusa MK3S MMU2S profile
* Update all filament profiles
* Drag model into buildplate
* Use Place on Face function to reorient model (this is important, in other orientations the error does not occur)
* Slice model
* Crash

This orientation slices fine:
[image]

This orientation causes a crash during slicing.
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Unfortunately this is a paid model. The license doesn't allow me to share the file.

Crash dump:
[PrusaSlicer_2020-11-22-121004_Jaaps-MacBook-Pro.crash.zip](https://github.com/prusa3d/PrusaSlicer/files/5579352/PrusaSlicer_2020-11-22-121004_Jaaps-MacBook-Pro.crash.zip)

audiorazor, 2020-11-22:
I am having the exact same issue.  Try to slice an mmu model and the slicer crashes.

drelich, 2020-11-22:
I don't even have to slice an mmu model. Simple one colour Deadpool bust crashes PrusaSlicer mid slicing... Looks like I'm going back to Alpha3.

TheLion, 2020-11-22:
I have the same issue on Windows.
It doesn't happen with every model though. When I add a particular modem to the build plate, no matter how muich more is on there, the slicing crashes.
I can share the STL's of needed, but they are part of a paid model (Steering Wheel A.stl and Steering Wheel B.stl) from the 3DSets Mode 6 Rancher 4x4 Wagon v1.0.2.

kgolger, 2020-11-22:
Same here, alpha4 crashes on several simple parts when layerhigh equal or greater then 0.1mm, with or without multimaterial.
[crashing_2.3.0_alpha4.zip](https://github.com/prusa3d/PrusaSlicer/files/5580150/crashing_2.3.0_alpha4.zip)

Gacky2k, 2020-11-23:
Yup...crashing when trying to slice even simple files.  Here is the offending file for me.  Printing a Minecraft figure for my grandson, and this is the helmet he can wear.  I'll attach the project file.  
[Minecraft_Steve-Helmet.zip](https://github.com/prusa3d/PrusaSlicer/files/5580738/Minecraft_Steve-Helmet.zip)

picturethis20, 2020-11-23:
Dealing with the similar issue plus a odd issue. First if I hit "
command + "I" on my keyboard it doesn't open up the  import stl window like in previous versions it just opens up a whole new prusaslicer app like opening up a new window in your browser. Then it crashes when I click on slice now. I'm hoping this is an easy fix for them because I was looking forward to all the new features. Till then I'll keep using Alpha3.

bubnikv, 2020-11-23:
Likely a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5208

jaapp, 2020-11-23:
I can confirm that trying to slice the Minecraft_Steve-Helmet.zip file shared by @Gacky2k also crashes PrusaSlicer during slicing

[image]

Crashdump:
[PrusaSlicer_2020-11-23-100029_Jaaps-MacBook-Pro.crash.zip](https://github.com/prusa3d/PrusaSlicer/files/5581844/PrusaSlicer_2020-11-23-100029_Jaaps-MacBook-Pro.crash.zip)

foreachthing, 2020-11-23:
> Yup...crashing when trying to slice even simple files. Here is the offending file for me. Printing a Minecraft figure for my grandson, and this is the helmet he can wear. I'll attach the project [...]

It only crashes if I import your entire 3mf. If I delete the config files (inside 3mf) then it works. So, my guess is, that your config does something funny.
[image]

Delete *.config in `Minecraft_Steve-Helmet.3mf\Metadata\`.

jaapp, 2020-11-23:
> It only crashes if I import your entire 3mf. If I delete the config files (inside 3mf) then it works. So, my guess is, that your config does something funny.
>
> Delete *.config in [...]

I got it to crash using the default profiles as wel. To double check I also extracted the STL from the 3MF and made a new project trying to slice the STL. Also resulted in a crash.

sej7278, 2020-11-23:
same here on linux - 2.3.0a4 segfaults so much more than a3 and 2.2.0 never crashed once.

it seems to be to do with scaling. if i load [marvin](https://www.thingiverse.com/thing:215703) and just click slice with my default settings it works fine.

if i scale it any larger than 117% it segfaults and i get this from gdb:

```gdb
[New Thread 0x7fff71ff7700 (LWP 1659114)]
[Thread 0x7fffd1bfc700 (LWP 1659046) exited]
--Type <RET> for more, q to quit, c to continue without paging--c

Thread 24 "slic3r_BgSlcPcs" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffeefde700 (LWP 1659083)]
0x00000000007a6a9a in Slic3r::mark_boundary_segments_touching_infill(std::vector<std::vector<Slic3r::Point, std::allocator<Slic3r::Point> >, std::allocator<std::vector<Slic3r::Point, std::allocator<Slic3r::Point> > > > const&, std::vector<std::vector<float, std::allocator<float> >, std::allocator<std::vector<float, std::allocator<float> > > > const&, std::vector<std::vector<Slic3r::ContourIntersectionPoint*, std::allocator<Slic3r::ContourIntersectionPoint*> >, std::allocator<std::vector<Slic3r::ContourIntersectionPoint*, std::allocator<Slic3r::ContourIntersectionPoint*> > > >&, Slic3r::BoundingBox const&, std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> > const&, double, double)::Visitor::operator()(int, int) ()
```

sometimes i get this error too:

```
[2020-11-23 12:54:51.636211] [0x00007fffeeffd700] [error]   GCodeProcessor encountered an invalid toolchange, maybe from a custom gcode.
```

Area5142, 2020-11-23:
In MMU2S mode, loads go fine, but color change of part makes a4 crash.
I have switched back to a3 now...

CWStevens, 2020-11-23:
[ABS_or_PETG-_Electronics_body_top_layer-4.zip](https://github.com/prusa3d/PrusaSlicer/files/5584333/ABS_or_PETG-_Electronics_body_top_layer-4.zip)
Crashed when slicing the attached .STL file

Revilo91, 2020-11-23:
Crashed when slicing with different infill settings:
[image]

Tested with this stl:
[deo-stand.zip](https://github.com/prusa3d/PrusaSlicer/files/5585269/deo-stand.zip)

foreachthing, 2020-11-23:
@Revilo91
> Crashed when slicing with different infill settings:

Got crashes with 'Archimedean Chord' and 'Octagram Spiral'.
No crash with Gyroid, though.

bubnikv, 2020-11-24:
Fixed with 62bdc192d87a9a1835e6ff6ff10b5a0893279c98

closed 2020-11-24 by bubnikv